nasa/cFE #3: Service not logging under Systemd

Immortalin, 2017-09-20:
When ran on POSIX systems with SystemD, cFE's log doesn't show up on `journalctl`, the default logging service.  It records all stdout by default. A test with a standard hello world C program on infinite loop showed that the stdout wasn't being buffered or detached. Please advise

Immortalin, 2017-09-21:
@the-other-james ?

the-other-james, 2017-09-21:
Hi @Immortalin,
I'm not sure what the issue is here, but maybe I'm out of my depth on this one (I am not very familiar with systemd, or logging in cFE for that matter). I'm pretty sure that logs in cFE are written to files as specified in: https://github.com/nasa/cFE/blob/master/cfe/fsw/platform_inc/cpu1/cfe_platform_cfg.h

The Linux build was originally made to enable easier testing of cFE: developers could get up and running quickly on their local machine instead of messing with a dev board. So I am not surprised if cFE does not take advantage of many features available in Linux. For example, cFE uses threads for applications instead of processes (the multi-threading model matches cFE's original targets: embedded real time operating systems like VxWorks and RTEMS).

Maybe the community site (http://coreflightsystem.org/) would be a better place to ask this question?

EzraBrooks, 2017-09-21:
Defining different logging behavior per OS should be doable via PSP/OSAL, although the current log layer isn't build to support it. Replacing logfile write calls in certain platforms with a journal write call would be the way to do it.

Immortalin, 2017-09-21:
When ran from shell, cFE prints to stdout. Journalctl usually captures everything that goes to stdout transparently i.e. no config needed but in this case it doesn't seem to be working.

skliper, 2019-04-18:
https://coreflightsystem.org/ has links to submit bugs or contact the cFS community for support if the issue still persists.

closed 2019-04-18 by skliper