nasa/cFE #4: Make Config fails

astro-gokart, 2017-09-26:
make config returns the following error:
```
>>> [PRE-BUILD] GET MISSING HEADER FILES FROM <CFS_MISSION> CORE <<<

mkdir: cannot create directory ‘inc’: File exists
make: *** [cfs_copy_core_incs] Error 1
```

`ls -al`
```
[tdev@localhost cpu1]$ ls -al inc 
-rw-rw-r--. 1 tdev tdev 1353 Sep 26 14:52 inc
```

`file` returns
```[tdev@localhost cpu1]$ file inc
inc: C source, ASCII text
```

```
GSFC Classic Build Version 6.5.0
>>> [PRE-BUILD] VERIFY OSAL CONFIGURED <<<
>>> [PRE-BUILD] GET MISSING HEADER FILES FROM <CFS_APP_SRC>/INC <<<

>>> [ENVS] MAKEFILE ENVIRONMENT VARIABLES <<<

MISSION = CFE_TST
‘/home/tdev/workspace/cFE/apps/inc/sample_lib.h’ -> ‘/home/tdev/workspace/cFE/build/mission_inc/sample_lib.h’

THE_APPS = sample_lib sample_app ci_lab to_lab sch_lab
>>> [PRE-BUILD] GET MISSING HEADER FILES FROM <APP>/PUBLIC_INC <<<
THE_TBLS = 

CFS_MISSION = /home/tdev/workspace/cFE

CFS_MISSION_INC = /home/tdev/workspace/cFE/build/mission_inc
>>> [PRE-BUILD] GET MISSING HEADER FILES FROM <APP>/MISSION_INC <<<
CFS_APP_SRC = /home/tdev/workspace/cFE/apps

CFE_FSW = /home/tdev/workspace/cFE/cfe/fsw
CFE_CORE_SRC = /home/tdev/workspace/cFE/cfe/fsw/cfe-core/src

CFE_CORE_PLATFORM_INC_SRC = cpu1
‘/home/tdev/workspace/cFE/apps/sample_app/fsw/mission_inc/sample_app_perfids.h’ -> ‘/home/tdev/workspace/cFE/build/mission_inc/sample_app_perfids.h’
>>> [PRE-BUILD] GET MISSING HEADER FILES FROM <APP>/PLATFORM_INC <<<
CFE_TOOLS = /home/tdev/workspace/cFE/tools
‘/home/tdev/workspace/cFE/apps/ci_lab/fsw/mission_inc/ci_lab_perfids.h’ -> ‘/home/tdev/workspace/cFE/build/mission_inc/ci_lab_perfids.h’

‘/home/tdev/workspace/cFE/apps/to_lab/fsw/mission_inc/to_lab_perfids.h’ -> ‘/home/tdev/workspace/cFE/build/mission_inc/to_lab_perfids.h’
‘/home/tdev/workspace/cFE/apps/sch_lab/fsw/mission_inc/sch_lab_perfids.h’ -> ‘/home/tdev/workspace/cFE/build/mission_inc/sch_lab_perfids.h’
CFE_PSP_SRC = /home/tdev/workspace/cFE/psp/fsw
OSAL = /home/tdev/workspace/cFE/osal
OSAL_SRC = /home/tdev/workspace/cFE/osal/src/os

BUILD_PATH = .
CFS_EXE = ./exe

CFS_PROM = ./exe/cf
>>> [PRE-BUILD] CREATING MISSING <APP> BUILD DIRECTORIES <<<
CFS_PROM_APPS = ./exe/cf/apps

CFS_RAM = ./exe/ram
>>> [PRE-BUILD] GET MISSING MAKEFILES FROM <APP>/FOR_BUILD <<<

‘/home/tdev/workspace/cFE/apps/sample_lib/fsw/for_build/Makefile’ -> ‘sample_lib/Makefile’
>>> [PRE-BUILD] CREATE MISSING TOOL BUILD DIRECTORIES <<<
‘/home/tdev/workspace/cFE/apps/sample_app/fsw/for_build/Makefile’ -> ‘sample_app/Makefile’

‘/home/tdev/workspace/cFE/apps/ci_lab/fsw/for_build/Makefile’ -> ‘ci_lab/Makefile’
‘/home/tdev/workspace/cFE/apps/to_lab/fsw/for_build/Makefile’ -> ‘to_lab/Makefile’
‘/home/tdev/workspace/cFE/tools/elf2cfetbl/for_build/Makefile’ -> ‘elf2cfetbl/Makefile’
‘/home/tdev/workspace/cFE/apps/sch_lab/fsw/for_build/Makefile’ -> ‘sch_lab/Makefile’

>>> [PRE-BUILD] GET MISSING HEADER FILES FROM <CFS_MISSION> CORE <<<

mkdir: cannot create directory ‘inc’: File exists
make: *** [cfs_copy_core_incs] Error 1
```

astro-gokart, 2017-09-26:
resolved by running `mkdir inc` before running `make config`

EzraBrooks, 2017-09-27:
I know this isn't a solution, but - are you intending to use the classic build system? The Makefiles are included for back-compatibility with existing cFE projects, I believe, whereas the CMake files are meant to be used going forward.

astro-gokart, 2017-09-27:
Good point; however, I was going off the readme.md doc 

that outlines the build steps like so:
```
cd cFE
. ./setvars
cd build/cpu1
make config
make
cd exe
sudo ./core-linux.bin```

Immortalin, 2017-10-02:
@astro-gokart if you have difficulty building, you can use the distro with a [prebuilt version integrated into Linux that I made](https://github.com/Immortalin/Perihelion_OS).

jcmarsh, 2017-10-10:
@astro-gokart I'm looking into this to see if I need to update the README.md, and I wasn't able to recreate your issue (`make config` creates the `inc` directory for me). What is your OS? Thanks!

aogrcs, 2018-01-04:
I had also the problem
>>> [PRE-BUILD] VERIFY OSAL CONFIGURED <<<

make[1]: *** /build: No such file or directory.  Stop.
../cfs.mak:51: recipe for target '/build/inc/osconfig.h' failed
make: *** [/build/inc/osconfig.h] Error 2

the-other-james, 2018-01-04:
I wasn't able to recreate this issue. On a clean install, the osal Makefile should test to see if 'osal/build/inc' exists and create it if not: https://github.com/nasa/osal/blob/75beb864db7913bc29dc0b2cdfb193e2002fe40b/build/Makefile#L45

Could you provide more details about your problem? What OS are you using?

For a quick fix, you can create the missing `osal/build/inc` directory and rerun `make`.

This is the output I get on Ubuntu 16.04:

    >>> [PRE-BUILD] VERIFY OSAL CONFIGURED <<<

    make[1]: Entering directory '/home/jcmarsh/test_cFE/osal/build'
    test -d inc || mkdir inc
    cp -f ../src/bsp/pc-linux/config/osconfig.h inc/osconfig.h
    make[1]: Leaving directory '/home/jcmarsh/test_cFE/osal/build'

aogrcs, 2018-01-05:
I used ubuntu 16.04, did you need other information? Thanks

aogrcs, 2018-01-05:
I got the problem! Substituted . ./setvars in the readme file with source setvars.sh

Sstewart26, 2018-11-20:
I have hit this problem with the Makefile system on a couple OS's.  CentOS, Ubuntu 16.04 32-bit and 18.04 64 bit.  In copying headers to the inc folder, it instead creates a file called inc which has the contents of the headers attempting to be copied.  This can be avoided by creating an inc folder before building but is a symptom of a larger issue.  Building with inc folder, the build fails because the it is trying to move compiled objects into folders that havent been created yet.  If you make twice, this gets further but still does not result in a built configuration that will run successfully.  It would be helpful if the Wiki were updated to also describe how the new CMake build system is supposed to be used.

the-other-james, 2018-11-20:
The latest version of cFS (6.6) converts over completely to using cmake, which should resolve this issue. Unfortunately, the public release of 6.6 has not yet been approved.

skliper, 2019-04-18:
https://coreflightsystem.org/ has links to submit bugs or contact the cFS community for support if the issue still persists.

closed 2019-04-18 by skliper