nasa/cFE #7: Need default included mission configuration for unit tests

TimJSwan89, 2017-12-06:
You can run the following as a script if you find that easier:
```
    git clone https://github.com/nasa/cFE.git
    cd CFE/
    git checkout cf6f7a771c52fa97c834e765a6c1cb50205c6df0
    rm -rf osal/
    git clone https://github.com/nasa/osal.git
    cd osal/
    git checkout 7139592f04e47f7522b07b1ef9f84a21393df88a
    cd src/os
    ln -s posix posix-ng
    cd ../../..
    cp -r cfe/cmake/sample_defs/ cfe/sample_defs/
    mkdir build-sim/
    cd build-sim/
    export SIMULATION=native
    cmake -DCMAKE_C_COMPILER=gcc -DENABLE_UNIT_TESTS=TRUE ../cfe
```
Then it complains about needing a `MISSIONCONFIG` directory.
When I create an empty directory, `x`, for that, it complains about `x_defs/targets.cmake` not existing.
Is there a default mission config that I can use? I want to build this with a custom compiler, but I need a method to build it with `gcc` first and get unit tests running.

jcmarsh, 2017-12-06:
Would https://github.com/nasa/cFE/blob/master/cfe/cmake/sample_defs/targets.cmake work?

This may be better suited for a question here: http://coreflightsystem.org/questions/

TimJSwan89, 2017-12-07:
Someone from my organization helped me out some. Earlier, I was moving `sample_defs` into the `cfe` directory when I was supposed to be moving it into the top directory. You can close the issue.

closed 2017-12-28 by the-other-james