nasa/cFE #8: "Could not load cFE application" error -- yet `.so` file exists

ChristopherRabotin, 2017-12-22:
cFE refuses to load my custom app. I built it from the `sample_app` provided, and it compiles correctly. Moreover `make realclean && make config && make` will create the expected `lls_app.so` and place them in the expected locations (i.e. `build/cpu1/exe/cf/apps/lls_app.so` and `build/cpu1/exe/lls_app.so`). Furthermore, the header of these dynamic libraries confirms that they compiled for the correct target when compared to the other default libraries which load.

Any hint is greatly appreciated as I've been struggling with this issue for several long hours now.

Here's the relevant log:
```
1980-012-14:03:20.25303 ES Startup: Loading file: /cf/apps/lls_app.so, APP: LLS_APP
1980-012-14:03:20.25304 ES Startup: loading directly:/cf/apps/lls_app.so (LLS_APP) -70028.
1980-012-14:03:20.25322 ES Startup: Could not load cFE application file:/cf/apps/lls_app.so. EC = 0xFFFFFFFF
```

And the full boot-up log:
```
CFE_PSP: Default Reset Type = PO
CFE_PSP: Default Reset SubType = 1
CFE_PSP: Default CPU ID = 1
CFE_PSP: Default Spacecraft ID = 66
CFE_PSP: Default CPU Name: CPU1
CFE_PSP: Starting the cFE with a POWER ON reset.
CFE_PSP: Clearing out CFE CDS Shared memory segment.
CFE_PSP: Clearing out CFE Reset Shared memory segment.
CFE_PSP: Clearing out CFE User Reserved Shared memory segment.
2027-356-20:29:57.62589 POWER ON RESET due to Power Cycle (Power Cycle).
2027-356-20:29:57.62591 ES Startup: CFE_ES_Main in EARLY_INIT state
CFE_PSP: CFE_PSP_AttachExceptions Called
2027-356-20:29:57.62596 ES Startup: CFE_ES_Main entering CORE_STARTUP state
2027-356-20:29:57.62597 ES Startup: Starting Object Creation calls.
2027-356-20:29:57.62598 ES Startup: Calling CFE_ES_CDSEarlyInit
2027-356-20:29:57.62609 ES Startup: Calling CFE_EVS_EarlyInit
2027-356-20:29:57.62613 Event Log cleared following power-on reset
2027-356-20:29:57.62614 ES Startup: Calling CFE_SB_EarlyInit
2027-356-20:29:57.62627 ES Startup: Calling CFE_TIME_EarlyInit
1980-012-14:03:20.00000 ES Startup: Calling CFE_TBL_EarlyInit
1980-012-14:03:20.00023 ES Startup: Calling CFE_FS_EarlyInit
1980-012-14:03:20.00050 ES Startup: Core App: CFE_EVS created. App ID: 0
EVS Port1 66/1/CFE_EVS 14: No subscribers for MsgId 0x808,sender CFE_EVS
EVS Port1 66/1/CFE_EVS 1: cFE EVS Initialized. cFE Version 6.5.0.0
1980-012-14:03:20.05067 ES Startup: Core App: CFE_SB created. App ID: 1
1980-012-14:03:20.05072 SB:Registered 4 events for filtering
EVS Port1 66/1/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_SB
EVS Port1 66/1/CFE_SB 1: cFE SB Initialized
1980-012-14:03:20.10086 ES Startup: Core App: CFE_ES created. App ID: 2
EVS Port1 66/1/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_ES
EVS Port1 66/1/CFE_ES 1: cFE ES Initialized
EVS Port1 66/1/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_ES
EVS Port1 66/1/CFE_ES 2: Versions:cFE 6.5.0.0, OSAL 4.2.1.0, PSP 1.3.0.0, chksm 15267
1980-012-14:03:20.15104 ES Startup: Core App: CFE_TIME created. App ID: 3
EVS Port1 66/1/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_TIME
EVS Port1 66/1/CFE_TIME 1: cFE TIME Initialized
1980-012-14:03:20.20121 ES Startup: Core App: CFE_TBL created. App ID: 4
EVS Port1 66/1/CFE_TBL 1: cFE TBL Initialized.  cFE Version 6.5.0.0
1980-012-14:03:20.25129 ES Startup: Finished ES CreateObject table entries.
1980-012-14:03:20.25135 ES Startup: CFE_ES_Main entering CORE_READY state
1980-012-14:03:20.25140 ES Startup: Opened ES App Startup file: /cf/apps/cfe_es_startup.scr
1980-012-14:03:20.25158 ES Startup: Loading file: /cf/apps/ci_lab.so, APP: CI_LAB_APP
1980-012-14:03:20.25160 ES Startup: loading directly:/cf/apps/ci_lab.so (CI_LAB_APP) -70028.
1980-012-14:03:20.25194 ES Startup: CI_LAB_APP loaded and created
EVS Port1 66/1/CI_LAB_APP 6: CI: RESET command
EVS Port1 66/1/CI_LAB_APP 3: CI Lab Initialized.  Version 2.2.0.0
1980-012-14:03:20.25217 ES Startup: Loading file: /cf/apps/to_lab.so, APP: TO_LAB_APP
1980-012-14:03:20.25219 ES Startup: loading directly:/cf/apps/to_lab.so (TO_LAB_APP) -70028.
1980-012-14:03:20.25245 ES Startup: TO_LAB_APP loaded and created
EVS Port1 66/1/TO_LAB_APP 1: TO Lab Initialized. Version 2.2.0.0 Awaiting enable command.
1980-012-14:03:20.25266 ES Startup: Loading file: /cf/apps/sch_lab.so, APP: SCH_LAB_APP
1980-012-14:03:20.25267 ES Startup: loading directly:/cf/apps/sch_lab.so (SCH_LAB_APP) -70028.
1980-012-14:03:20.25286 ES Startup: SCH_LAB_APP loaded and created
SCH Lab Initialized.  Version 2.2.0.0
1980-012-14:03:20.25303 ES Startup: Loading file: /cf/apps/lls_app.so, APP: LLS_APP
1980-012-14:03:20.25304 ES Startup: loading directly:/cf/apps/lls_app.so (LLS_APP) -70028.
1980-012-14:03:20.25322 ES Startup: Could not load cFE application file:/cf/apps/lls_app.so. EC = 0xFFFFFFFF
1980-012-14:03:20.25324 ES Startup: CFE_ES_Main entering OPERATIONAL state
EVS Port1 66/1/CFE_TIME 21: Stop FLYWHEEL
```

the-other-james, 2017-12-22:
Please direct your questions here: http://coreflightsystem.org/questions/
You may want to also post your startup script and double check the name of the entry function.

ChristopherRabotin, 2017-12-22:
[X-POST](http://coreflightsystem.org/questions/question/could-not-load-cfe-application-error-yet-so-file-exists-8)

I'm told by a coworker this error is typical of a runtime error when loading the app. I'll investigate that issue.

ChristopherRabotin, 2017-12-22:
That error code does indeed mean the app itself encountered a runtime error.

In my case, I had a typo in one of the functions called from in the `*_AppInit` initialization function.

closed 2017-12-22 by ChristopherRabotin