nasa/cFE #16: About the PSP mak file

aogrcs, 2018-12-03:
Hi
I'm very new for write .mak file. I checked gcc(5.5.0) manual, found some symbols such as -Wall, -m32, but did not find __ix86__. Could you tell me how can I find this? Which manual should I reference to? I want to make psp and osal for raspberry or freertos! Thanks
Sincerely

the-other-james, 2018-12-04:
The compiler flags you need will depend on how you intend to compile your system. If you cross-compile (for example, compile cFS / FreeRTOS on your personal computer and then transfer it to the Raspberry Pi), you will need to install a toolchain for your target (Raspberry Pi works with 32 or 64 bit ARM).

I'm not sure what you are looking for with "ix86".

aogrcs, 2018-12-05:
Thanks, James

closed 2018-12-05 by aogrcs