nasa/cFE #17: Support for OSX

SweeWarman, 2019-04-17:
Is the cFS version available on the master branch compatible with OSX? Looking through the files, it seems like cFS uses a lot of the extended posix functionalities (e.g. timers, thread prioritization and certain real time signals) that are not available on OSX. Although OSX is posix compliant, it doesn't support extended posix functionalities. 

From a development standpoint, support for OSX is desirable. I have developed some patches that can be applied to some osal and psp src files to enable compilation on OSX (including a reimplementation of the timer API using the grand central dispatch on OSX). If the cFS developers are interested in looking into this, I can integrate these patches and submit a pull request.

skliper, 2019-04-18:
https://coreflightsystem.org/ has links to submit bugs or contact the cFS community, and they will likely provide more input than via a GitHub issue.

There are open internal tickets (osal #180, psp #75) detailing some of the issues but I don't know of anyone actively working it.  There was a PSP submitted based on pc-linux, not sure the level of testing done on it.  Many folks end up using a virtual linux machine for development.

- There is no timer interface.
- pthreads does not include the priority control API's
- There are no "_init" or "_fini" externs for stack introspection.
- No vfs.h
- dlsym takes special RTLD_* handles to define its namespace search behavior (NULL does not work.)

skliper, 2019-05-31:
Heads up that the cFS Framework (and associated community configuration control board) is scoped to cover the linux, RTEMS, and VxWorks implementations of OSAL.  We currently aren't adding additional OS's to the framework, but we encourage users to create project and submit a link to the https://github.com/nasa/cfs README.md for other operating systems they have an implementation for.

closed 2019-05-31 by skliper