nasa/cFE #20: Incorrect SCID field length

CutrightD, 2019-08-05:
cfe_mission_cfg.h shows a note circa line 49 that 'CCSDS allocates 8 bits for this field in the standard VCDU'. Newer CCSDS versions actually allocate 10-bits for this field - this seems like a minor thing but could potentially have implications elsewhere in the code. Per @acudmore , maybe TO, DS, and CF?

skliper, 2019-08-12:
Note fsw/mission_inc/cfe_mission_cfg.h is only used in the deprecated classic build.  CMake uses the mission configuration in the top level sample_defs, which is typically copied from the cFE/cmake directory.

https://github.com/nasa/cFE/blob/8811c92094512975e14fd545df355705141560b2/cmake/sample_defs/sample_mission_cfg.h#L47-L50

only real reference is from CFE_SB_SetMsgId:

https://github.com/nasa/cFE/blob/8811c92094512975e14fd545df355705141560b2/fsw/cfe-core/src/sb/cfe_sb_msg_id_util.c#L208

Structure is defined as:

https://github.com/nasa/cFE/blob/8811c92094512975e14fd545df355705141560b2/fsw/cfe-core/src/inc/ccsds.h#L140-L154

And write implemented as:

https://github.com/nasa/cFE/blob/8811c92094512975e14fd545df355705141560b2/fsw/cfe-core/src/inc/ccsds.h#L353-L354

and copies all 16 bits, so no impact to cFE.  Could update/remove the 8 bit field comment.

Any impacts to TO, DS, or CF would need to be addressed in their repos (not a cFE issue).

skliper, 2019-09-16:
No change applicable to cFE, closing ticket here.

closed 2019-09-16 by skliper