nasa/cFE #120: ES Does Not Check CFE_PSP_MemRead8 Return Code

skliper, 2019-09-30:
The ES API functions need to properly alert users of errors.  The CFE_ES_CalculateCRC API function will not alert users if an error code was returned from the call to the CFE_PSP_MemRead8.

skliper, 2019-09-30:
_Trac comment by jphickey on 2015-12-08 11:59:50:_

In the current development branch, in cfe_es_api.c, `CFE_ES_CalculateCRC()` function, the code is:

{{{
for ( i = 0 ; i < DataLength ; i++,  BufPtr++)
{
    if (CFE_PSP_MemRead8((cpuaddr)BufPtr, &ByteValue) != CFE_PSP_SUCCESS)
    {
        ByteValue = 0;
    }
    Index = ( ( Crc ^ ByteValue) & 0x00FF);
    Crc = ( (Crc >> 8 ) & 0x00FF) ^ CrcTable[Index];
}
}}}

The return code of CFE_PSP_MemRead8 is indeed being checked, but it is handled silently, not reporting this error to the caller.

The API returns the CRC directly, and therefore does not have a provision for returning an error.  To report this to the caller would require an API change which would in turn break backward compatibility.  

Since CFE_PSP_MemRead8 is extremely unlikely to fail, since it only operates on local RAM on any platform we actually have implemented thus far.  Therefore I am not sure this meets the cost/benefit trade off of a breaking API change.

skliper, 2019-09-30:
_Trac comment by glimes on 2016-10-18 15:07:03:_

Remove my name from many tickets (where it was applied by default), so others can tell that there are tickets that need some attention.

skliper, 2019-11-04:
https://github.com/nasa/PSP/issues/10 - CFE_PSP_MemRead8 is targeted for deprecation.

jphickey, 2019-11-04:
Note that the current master branch no longer has the API call in question here.  It was removed prior to github migration, and replaced with a direct memory read.  It is no longer calling `CFE_PSP_MemRead8` at all.

Therefore I would recommend closing this as invalid, regardless with what happens with nasa/PSP#10, as this is no longer an issue in CFE_ES either way.

closed 2019-11-04 by skliper