nasa/cFE #123: cFE Time subsystem has calls to OS functions that do not exist

skliper, 2019-09-30:
In the cFE Time subsystem, there are a couple of calls to functions that do not exist: OS_GetLocalMET and OS_SetLocalMET. These calls are conditionally compiled by a configuration parameter, and probably not used often.

Note: GSFC's MMS FSW uses these functions and had to provide them.

skliper, 2019-09-30:
_Trac comment by jphickey on 2015-08-31 11:50:01:_

See also #78

skliper, 2019-09-30:
_Trac comment by glimes on 2015-09-02 13:05:01:_

It is perhaps telling that the ONLY mentions of these two
functions (that I can find) are in the specific places in
CFE_TIME that are commented as interacting with a hardware
register.

I rather expected to find an "extern" function prototype
either in OSAL or in PSP to nail down this external API.

As with #78 this is contingent on a configuration option,
and the calls for these look like this:

{{{
    /*
    ** Maintain virtual MET as count read from h/w MET register...
    */
#if (CFE_TIME_CFG_VIRTUAL != TRUE)
    OS_GetLocalMET(&CFE_TIME_TaskData.VirtualMET);
#endif
}}}

(The other two call sites are similar)

I suspect that fixing this is something that actually happens
over in the PSP area. Do we need a ticket in CFE_PSP to track?

skliper, 2019-09-30:
_Trac comment by sstrege on 2016-08-26 14:08:10:_

This points to a much larger issue of the need to overhaul Time Services.  This is a feature all missions/projects have been living with unaffected.  This ticket has been deferred as it is not a high priority.

skliper, 2019-09-30:
_Trac comment by glimes on 2016-10-18 15:07:03:_

Remove my name from many tickets (where it was applied by default), so others can tell that there are tickets that need some attention.

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-04-01 09:41:37:_

CCB 3/27/2019 - discussed removing unimplemented configuration options.  Assigning to Alan for review.

Edit - fixed CCB date typo

skliper, 2019-09-30:
_Trac comment by acudmore on 2019-04-03 11:12:40:_

For the Dellingr Cubesat, we had CFE_TIME_CFG_VIRTUAL set to false, so I had to implement OS_GetLocalMET and OS_SetLocalMET. 

I would like to do a survey of recent missions and find out if any other users needed these functions. 

If we decide these functions are necessary for some users, I would recommend moving them to PSP functions, since mission time hardware is not a generic OS function. 

However, before adding any PSP functions, we may want to take a step back and consider simplifying the TIME subsystem.

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-06-24 15:53:01:_

Related to [cfs_psp:71], pending TIME subsystem design review

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-07-03 12:58:32:_

Moved unfinished 6.7.0 tickets to next release

skliper, 2020-02-26:
Related to #302

mlouielu, 2020-04-14:
Hi @skliper, so in your scenario, the flight board has another specific high-resolution clock for the MET, not only the the one such as `clock_gettime` in Linux?

skliper, 2020-04-14:
HI @mlouielu - Yes.  Frequently timing requirements are such that hardware is involved, both for internal time keeping and time management across a system.