nasa/cFE #127: Add support to allow SBN to pass sender information across the network (GSFC DCR 22063)

skliper, 2019-09-30:
A cFE interface is needed to allow SBN to pass sender information (ProcessorID, Application Name) across a network.
 
One proposed change is a new private function called CFE_SB_PassMsgWithSenderID. This function will be needed by SBN ONLY. The CCB should discuss how this function can be used by SBN and blocked from being called/used by other applications. Analysis is needed to determine the best solution.

skliper, 2019-09-30:
_Trac comment by cdknight on 2016-07-12 15:39:20:_

Currently SBN-injected messages have the application name of "SBN"--which SBN then uses to ensure that it ignores messages with the sender of "SBN", preventing loops. If the above functionality is approved and adopted, some other mechanism will be needed to ensure that SBN can identify which messages it generated versus which messages are from other applications in the cFE.

skliper, 2019-09-30:
_Trac comment by glimes on 2016-10-18 15:07:03:_

Remove my name from many tickets (where it was applied by default), so others can tell that there are tickets that need some attention.

skliper, 2019-11-04:
See also https://github.com/nasa/cFE/issues/292

skliper, 2020-02-26:
@CDKnightNASA is this enhancement request still valid?  It hasn't been updated in years, so I'm wondering if it's OBE.

CDKnightNASA, 2020-10-01:
In researching this, I noticed there's a SB global variable "SenderReporting", which is initialized but not referenced anywhere in cFE. I do think it would still be worth pursuing this, but I would need to give it some deep thought as it may be possible to determine by looking up the current task name.

skliper, 2020-10-01:
So one approach is to include the source information in the message, which is then a mission managed identifier that is unique.

CDKnightNASA, 2020-10-01:
> So one approach is to include the source information in the message (or message header), which is then a mission managed identifier that is unique.

True, I think this may actually be the preferred mechanism. I'm guessing that for > 90% of messages that are published, the publisher is known to the mission. In considering this, I will close this ticket.

closed 2020-10-01 by CDKnightNASA

jwilmot, 2020-10-01:
I know this is now  closed but this was part of SB and SBN in 2013. Not sure who took it out or why. It was intended for last sender, local and over SBN.  It was added DCR469:1 Added GetLastSender code ** Revision 1.68 2007/04/24 11:07:18EDT
From cfe_sb_api.c
  /* store the sender information */
    if(CFE_SB.SenderReporting != 0)
    {
       if (SenderPtr == NULL) { /* This is a normal SB send call */
          BufDscPtr->Sender.ProcessorId = CFE_PSP_GetProcessorId();
         strncpy(&BufDscPtr->Sender.AppName[0],CFE_SB_GetAppName(TskId,FullName),OS_MAX_API_NAME);
        }
       else { /* Caller wants to set the Sender metadata (Likely an SBN call)  */ 
          BufDscPtr->Sender.ProcessorId = SenderPtr->ProcessorId;
          strncpy(&BufDscPtr->Sender.AppName[0], &SenderPtr->AppName[0], OS_MAX_API_NAME);
        }
        
    }