nasa/cFE #139: TBL - Dump Table Registry Data Command Can Hog CPU (GSFC DCR 23031)

skliper, 2019-09-30:
The GSFC NICER FSW team discovered that it takes approximately 1 second to process the Dump Table Registry command when there are 250 RTS tables and that lower priority tasks generate various error events related to not getting an opportunity to run. Adding a platform configuration definition to insert task delays into the process loop will increase the time necessary for CFE_TBL to finish the command but will allow lower priority tasks to run while the command is being processed.

Also, creating some test code that puts performance markers around the file i/o calls within the command processor may identify where the bulk of the command process time is spent. It may be possible to reduce the command execution time by optimizing the command handler code if the bulk of the execution time is not being spent inside the file system.

skliper, 2019-09-30:
_Trac comment by glimes on 2016-10-18 15:07:03:_

Remove my name from many tickets (where it was applied by default), so others can tell that there are tickets that need some attention.

skliper, 2020-10-07:
The file writing should likely be moved to a low priority child task, same for any other potentially "large" file writes: CFE_SB_SendRtgInfo, CFE_SB_SendPipeInfo, CFE_SB_SendMapInfo

jphickey, 2020-10-08:
These all sound like great candidates for moving into the ES background job service.

Make a generic file writer job template based on the current perf data dump, then apply it for all these file dump commands.

skliper, 2020-10-13:
Note - standby on this until #928 is implemented (for SB writes), since they are being refactored.

skliper, 2021-01-06:
Removed dependency label, this can be resolved now.

PR #1148 (2021-02-17): Fix #139, Do file writes in background
**Describe the contribution**
Implement a generic FS facility to perform file writes as a background job.  

Applications wanting to use this facility need to instantiate a state object (metadata) in global memory, and two callback APIs- one to get a data record, another to send events. 

The following file requests are changed to use this facility:
- ES ER Log dump
- SB Pipe Info
- SB Message Map
- SB Route Info
- TBL Registry Dump

Fixes #139

**Testing performed**
First built and ran "main" branch (unchanged) and issued all file write commands before c [... cut]
commit 37df32d5 (2021-02-17): Merge pull request #1148 from jphickey/fix-139-filewrite-background

Fix #139, Do file writes in background
commit 81cbc46c (2021-02-17): Fix #139, move file writes to background task

Implements a generic asynchronous "file write request" facility
in the FS subsystem.  Given a metadata/control block with the file
details and writer state and appropriate callbacks,, this will
execute the file write job as part of the ES background task.

The following file requests are changed to use this facility:
- ES ER Log dump
- SB Pipe Info
- SB Message Map
- SB Route Info
- TBL Registry Dump
PR #1171 (2021-02-26): cFE Integration candidate: 2021-02-23
**Describe the contribution**
Fix #139, Do file writes in background
Fix #978, put task parameters into task record
Fix #790, allow directories and symlinks in FILELIST. 
Fix #735, add comment if null terminated or not.

Fix #1160, Shorten task info default filename 
Fix #1172, Update to use Ut_Stub_CheckDefaultReturnValue API  

**Testing performed**
See <https://github.com/nasa/cFS/pull/193/checks>

**Expected behavior changes**
PR #1148 - Implements a generic FS facility to perform file writes as a background job. Applications wanting to use this need t [... cut]
closed 2021-02-26 by astrogeco