nasa/cFE #189: EVS Unit Test Code Coverage Incomplete in Task.c (GSFC DCR 8492)

skliper, 2019-09-30:
cfe_evs_task.c - 99.61% % coverage (2 Lines of Code NOT executed)

The lines not covered are in CFE_EVS_TaskInit:

        -:  385:   /* Register EVS task for event services */
        5:  386:   Status = CFE_EVS_Register(NULL, 0, CFE_EVS_BINARY_FILTER);
        5:  387:   if (Status != CFE_SUCCESS)
        -:  388:   {
    #####:  389:      CFE_ES_WriteToSysLog("EVS:Call to CFE_EVS_Register Failed:RC=0x%08X\n",(unsigned int)Status);
    #####:  390:      return Status;
        -:  391:   }

REASON: None

skliper, 2019-09-30:
_Trac comment by glimes on 2016-10-18 15:07:03:_

Remove my name from many tickets (where it was applied by default), so others can tell that there are tickets that need some attention.

avan989, 2020-01-09:
I can do this.

skliper, 2020-01-09:
Excellent, I suggest start by rerunning the test to identify any currently missing coverage.

commit 067ffe23 (2020-01-21): Fix #189, Add test coverage for cfe_efs_task.c
commit 94f91a9d (2020-01-21): Merge pull request #464 from avan989/update_unitTest

Fix #189, Unit test for cfe_evs_task.c
PR #490 (2020-01-27): Integration Candidate 20200121
**Describe the contribution**
Fix #459, fix #355, fix, #24, fix #101, fix #412, fix #189

**Testing performed**
Steps taken to test the contribution:
1. Current CI passes
1. Enhanced CI in #40 passes debug/with deprecated elements testing

**Expected behavior changes**
- Resolve doxygen warnings
- add OMIT_DEPRECATED flag
- Add strict warnings and enforce
- Always increment sequence counter
- Fix comments
- Add code coverage test to evs

**System(s) tested on**
 - CI: Ubuntu 18.04

**Additional context**
N/A

**Third party code**
N/A

**Contributor Info - All inf [... cut]
closed 2020-01-27 by skliper