nasa/cFE #190: ES Unit Test Code Coverage Incomplete in apps.c

skliper, 2019-09-30:
cfe_es_apps.c - 98.23% coverage (8 Lines of Code NOT executed)

In CFE_ES_AppCreate:

-: 619: else /* Not a dynamic load - app must be statically linked into current exe */
-: 620: {
#4430987 621: StaticEntryPtr = (CFE_ES_AppStaticModuleApi_t *)EntryPointData;
#4430987 622: StartAddr = StaticEntryPtr->Ptrs.EntryFuncAddr;
#4430987 623: ModuleId = 0;
-: 624: }
23: 643: if (StaticEntryPtr != NULL)
-: 644: {
-: 645: /* Leave entry point / filename blank if static load entry */
#4430987 646: CFE_ES_Global.AppTable[i].StartParams.EntryPoint[0] = 0;
#4430987 647: CFE_ES_Global.AppTable[i].StartParams.FileName[0] = 0;
-: 648: }

In CFE_ES_LoadLibrary:

-: 932: else /* Not a dynamic load - library must be statically linked into current exe */
-: 933: {
#7210210 934: StaticEntryPtr = (CFE_ES_AppStaticModuleApi_t *)EntryPointData;
#7210210 935: FunctionPointer = StaticEntryPtr->Ptrs.LibInitFunc;
#7210210 936: ModuleId = 0;
-: 937: }

These lines of code can be covered in unit testing.

skliper, 2019-09-30:
_Trac comment by glimes on 2016-10-18 15:07:03:_

Remove my name from many tickets (where it was applied by default), so others can tell that there are tickets that need some attention.

avan989, 2020-01-09:
Original posting no longer valid. Current coverage for cfe_es_apps.c = 96.9 %
Missing coverage test for the following:

CFE_ES_AppCreate

```
    646           0 :                 CFE_ES_WriteToSysLog("ES Startup: Error removing temp RAM disk file, EC = 0x%08X\n",
     647             :                                      (unsigned int) ReturnCode);
```

CFE_ES_CleanupObjectCallback
```
    1363           0 :     case OS_OBJECT_TYPE_OS_MODULE:
    1364           0 :         Status = OS_ModuleUnload(ObjectId);
    1365           0 :         break;
    1366           0 :     default:
    1367           0 :         ObjIsValid = false;
    1368           0 :         Status = OS_ERROR;
    1369           0 :         break;
```

CFE_ES_CleanupTaskResources
`1467           0 :             Result = CFE_ES_APP_CLEANUP_ERR;`

CFE_ES_CountObjectCallback
 ```
   1495           0 : void CFE_ES_CountObjectCallback(uint32 ObjectId, void *arg)
    1496             : {
    1497             :     uint32                 *CountState;
    1498             :     uint32                 idtype;
    1499             : 
    1500           0 :     CountState = (uint32 *)arg;
    1501           0 :     idtype = OS_IdentifyObject(ObjectId);
    1502           0 :     if (idtype < OS_OBJECT_TYPE_USER)
    1503             :     {
    1504           0 :         ++CountState[idtype];
    1505             :     }
    1506           0 : }
```

skliper, 2020-01-09:
See #465 for recent coverage results

skliper, 2020-02-26:
OBE, closing

closed 2020-02-26 by skliper