nasa/cFE #202: cFE Application Developers Guide Should be Doxygen/Markdown Based

skliper, 2019-09-30:
As discussed at the 5-24-16 CCB meeting (reference cFS Systems Engineering Handbook discussion in the minutes), the cFE application developers guide contains a significant amount of API references.  Indirectly referencing the cFS API's in an external document is prone to error.  The API information should be referenced directly from the code.  It is highly recommended to move the Application Developers Guide to the cFE Doxygen User’s Guide.

skliper, 2019-09-30:
_Trac comment by sstrege on 2016-09-15 20:50:25:_

This guide should also include information on how to write endian neutral code:

On Thursday, September 15, 2016 7:10:10 PM EDT Scott, Michael S. (ARC-TI)[SGT, INC] wrote:
> In our project we used this:
> 
> INCLUDE(TestBigEndian)
> 
> TEST_BIG_ENDIAN(BIGENDIAN)
> IF(${BIGENDIAN})
>      ADD_DEFINITIONS(-DBIGENDIAN)
> ENDIF(${BIGENDIAN})
> 

This is the correct way to do it in cmake, but as Joe said there isn't really much reason to know this in cmake. 

With regard to Joe's email, a "better" way to do this in C is:

#define IS_LITTLE_ENDIAN (((union {unsigned x; char c;}){1}).c)

This is known at compile time, which is nice. So you can even do something like

#if IS_LITTLE_ENDIAN
...
#else
...
#endif

Steven

skliper, 2019-11-04:
Note #361 added a markdown version, but conversion to doxygen would make it linkable and reduce duplicated documentation.

jphickey, 2019-11-04:
> Note #361 added a markdown version, but conversion to doxygen would make it linkable and reduce duplicated documentation.

Can't the markdown file itself just be included in the doxygen?  Doxygen supports using markdown files and including that in the generated HTML, and terms should be cross-linked as expected.

skliper, 2019-11-04:
Absolutely, although I'd suggest another scrub in that context to remove duplicated documentation.