nasa/cFE #243: Continuation of EDS integration for CFE

skliper, 2019-09-30:
Integrate the remaining EDS features which are not yet included as part of the CFE 6.6 release.

This is a continuation of #207 for the next CFE release.

Defects to resolve: #92, TBD
Enhancements to resolve: TBD

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-03-20 15:39:09:_

Changed to a "task" ticket, and preference is to treat this as a container for all the tickets resolved by the EDS update and remaining work tickets.

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-07-03 12:58:32:_

Moved unfinished 6.7.0 tickets to next release

Jkillelea, 2022-07-24:
Hi there,

I wanted to ask about the status of EDS integration with cFS. As far as I know, there's a working version available from one of the flight software workshops with EDS integration fully involved in the pipeline, but I don't see any EDS tooling in the build process for mainline cFS. There are some XML documents in the cFE core, but no explanations about how to integrate EDS with user apps or ground stations.

skliper, 2022-07-25:
There's a few EDS/cFS related repos at https://github.com/jphickey/cfe-eds-framework, https://github.com/nasa/cFS-EDS-GroundStation, https://github.com/nasa/EdsLib.

Although further integration is a future goal, I'm not aware of any resources or funding on our side specifically targeting advancement of this goal with respect to the open-sourced software in the short term.  As always we encourage contributions, and feel free to reach out to the [cFS Product Team](https://github.com/nasa/cFS#contact-the-cfs-product-team) for more detailed collaboration discussions.  The discussions page on cFS is also a good resource for community coordination.