nasa/cFE #245: type safety

skliper, 2019-09-30:
In 6.6, as we move to supporting MsgId's, MsgKey's, RouteIdx, and other types, we should move away from using native C types and wrapping the types in a struct to prevent accidentally using the wrong type in assignments and function calls.

This will, of course, require re-tooling any existing code that expects the type to be a simple type...

For example, instead of:

{{{
typedef uint16 CFE_SB_MsgId_t;

uint32 SomeFunction(CFE_SB_MsgId_t MsgId) {
    /* ... */
    return MsgId; /* this will work fine, even though it's bad behavior! */
}
}}}

Use:

{{{
typedef uint16 CFE_SB_MsgId_Atom_t;
typedef struct
{
    CFE_SB_MsgId_Atom_t __MsgId; /* private, do not touch */
} CFE_SB_MsgId_t;

uint32 SomeFunction(CFE_SB_MsgId_t MsgId) {
    /* ... */
    return MsgId; /* will now give a compiler error */
}
}}}

skliper, 2019-09-30:
_Trac comment by cdknight on 2017-10-24 14:03:11:_

Note that the struct should read (sorry the double-underscore got interpreted by the trac system):

{{{
typedef struct
{
    CFE_SB_MsgId_Atom_t __MsgId; /* private, do not touch **/
} CFE_SB_MsgId_t;
}}}

skliper, 2019-09-30:
_Trac comment by cdknight on 2017-10-24 14:04:04:_

Curious what the performance impact will be for this, particularly for frequently-used types.

One option would be to have a compile-time flag that enables the wrappers so you can do compiler type checking, then disable for performance.

A-la:
{{{
#ifdef BUILD_WRAPPERS

typedef uint16 CFE_SB_MsgId_Atom_t;
typedef struct
{
    CFE_SB_MsgId_Atom_t __MsgId; /* private, do not touch */
}

#else

typedef uint16 CFE_SB_MsgId_t;

#endif
}}}

skliper, 2019-09-30:
_Trac comment by glimes on 2017-10-24 14:05:16:_

I have the power to edit the Description for you (this may
be one of those silly "admin only" things).

skliper, 2019-09-30:
_Trac comment by cdknight on 2017-10-24 14:09:44:_

Replying to [comment:4 glimes]:
> I have the power to edit the Description for you (this may
> be one of those silly "admin only" things).

yes, you silly admin. :D

skliper, 2019-09-30:
_Trac comment by glimes on 2017-10-24 14:10:46:_

Before you make the code complicated in the name of making it faster, please prototype and measure. You may find that there is no runtime cost.

Generally, I've found that
- copy structure by assignment
- pass structure to function by value
- return structure from function by value
tend to be just as fast as an int (or a double), for structures
that are as small as an int (or a double).

skliper, 2019-09-30:
_Trac comment by cdknight on 2017-10-24 14:11:47:_

Note also I filed ticket #241 suggesting creating a type for status/return codes so that it's clear when functions are returning a status. I have a first pass at the ES patch demonstrating the code change impact.

skliper, 2019-09-30:
_Trac comment by cdknight on 2017-10-24 15:00:38:_

whoop, not 6.6

skliper, 2019-09-30:
_Trac comment by cdknight on 2017-10-27 13:24:21:_

also note we may want to wrap void *'s wherever possible/needed (see ticket #247 for an example where we got bit by a void *.)

skliper, 2019-09-30:
_Trac comment by cdknight on 2017-10-27 14:26:57:_

see also ticket #172

skliper, 2019-09-30:
_Trac comment by jphickey on 2018-05-14 12:36:48:_

Upvote this ticket to be addressed in the next CFE release.

See also #100.  We should use the same approach for CFE Pipe IDs as I'm seeing some potential (mis)use of pipe IDs as array indices.

In every reasonable compiler out there, passing a struct by value is just as efficient as passing an integer of the same size (provided the structure size is less than or equal to `sizeof(long)`).  So wrapping an integer value in a struct should have no effect on performance at all.

I would advocate using this approach everywhere we have an "id" type that is numeric in nature.  It ensures that it cannot be mixed with other integer types haphazardly, and also helps preserve architectural integrity by forcing one to treat it as an opaque value, only going through the correct API to manipulate the values.

skliper, 2019-09-30:
_Trac comment by jphickey on 2019-03-27 11:23:51:_

Branch `trac-214-msgid-type-safety`, commit [changeset:d61f598] implements this concept for the `CFE_SB_MsgId_t` type. 

This was originally implemented in #206 but not included in CFE 6.6 due to the API compatibility concerns at the time of review.  I have rebased it to the current development branch and associated it with this ticket instead.

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-07-03 12:58:32:_

Moved unfinished 6.7.0 tickets to next release

jphickey, 2020-04-06:
Submitted tickets and pull requests in all app repos to be compatible going forward:
- nasa/sample_app#53
- nasa/ci_lab#45
- nasa/to_lab#28
- nasa/sch_lab#37

astrogeco, 2020-06-01:
@skliper I was having a conversation about message IDs and remembered this ticket. I think based on the discussion in https://github.com/nasa/cFE/pull/592 and https://github.com/nasa/cFE/pull/602 this issue shouldn't have been closed.

skliper, 2020-06-01:
@astrogeco - right.  Only partially addressed.

commit 3b7f20fb (2020-04-22): Merge pull request #602 and #635

Manually merged to resolve conflicts.

From branch jphickey/fix-263-msgid-api:

Fix #245, Consistent use of MsgId type
Fix #263, Expose CFE_SB_IsValidMsgId() API

From branch dmknutsen/issue_240:

Fixes #240, removes MESSAGE_FORMAT_IS_CCSDS ifdefs from code
commit 51c599c6 (2020-04-22): Fix #245, Consistent use of MsgId type

Make CFE core apps consistent in their use of the CFE_SB_MsgId type
and with the CFE SB API.  This employs the CFE SB API whenever any
of the following needs to happen:

- Use of a CFE_SB_MsgId_t value within a printf (event, syslog, etc).
- Initialization of a CFE_SB_MsgId_t from an integer value
- Comparison of two CFE_SB_MsgId_t values
- Checking if a CFE_SB_MsgId_t value is within the valid set

A few new macros are introduced, mainly because the inline functions
that already existed for this purpose cannot be used where evaluation
must be done at co [... cut]
PR #2001 (2021-12-06): cFE Integration Candidate, Caelum+dev1
**Describe the contribution**

PR #1663
- Fix #1546, use cfe time clock state enum t in cmd/tlm and handling

PR #1974
- Fix #1945, add CFE_SB_ValueToMsgId/MsgIdToValue wrappers
- Fix #1944, use 0 as the invalid CFE_SB_MsgId_t value
- Fix #245, use type safe value for CFE_SB_MsgId_t 

PR #1973 
- Fix #1818, EVS functional test add for different event types

PR #1954
- Fix #1952, Patch for recursive event loop
 
PR #1911
- Fix #1300, CFE_ES_MemAddOff structure to associated CFE_ES_MemAddress_t and CFE_ES_MemOffset_t

PR #1986
- Fix #1984, Avoid aliasing wa [... cut]
PR #1975 (2021-12-06): Fix #245, #1944, Message ID type improvements
**Describe the contribution**

1. Makes the value for `CFE_SB_MSGID_RESERVED` and `CFE_SB_INVALID_MSG_ID` to be based on 0 rather than -1.  This way, objects/values which are memset or otherwise cleared to 0 will get a safe value by default, rather than actively needing to be set to a safe value.
2. Makes the type used for `CFE_SB_MsgId_t` to be a wrapper around the MsgId value, rather than a direct equivalent of the MsgId value.  This increases type safety and enforces that the correct type and correct conversions (where necessary) must be employe [... cut]
closed 2021-12-06 by astrogeco