nasa/cFE #275: ES comments need checked for consistency

skliper, 2019-09-30:
Lots of copy/paste type errors in ES. For example, after CFE_ES_IncrementTaskCounter() the comment says  /* End of CFE_ES_ExitChildTask() */

skliper, 2019-09-30:
_Imported from trac issue 244.  Created by sseeger on 2018-08-14T19:37:54, last modified: 2019-07-03T12:48:08_

skliper, 2019-11-17:
Just piling on, `/* end of file */` in the middle of the file and a handful of others.

skliper, 2020-05-05:
Comments and format cleanup planned once this round of PRs are closed

thnkslprpt, 2022-10-27:
Most or all of the typos/mistakes that were originally noticed when this issue was opened have probably been cleared up by now.

I would suggest that this can be closed as completed by recent commits.