nasa/cFE #288: SB: Remove PipeName from CFE_SB_PipeD_t

skliper, 2019-09-30:
PipeName in CFE_SB_PipeD_t is redundant with the OSAL queue name and should be removed.

skliper, 2019-09-30:
_Trac comment by cdknight on 2018-10-23 22:15:23:_

See also ticket #287

skliper, 2019-09-30:
_Trac comment by cdknight on 2018-10-24 12:49:09:_

change committed [changeset:a8d1216]

skliper, 2019-09-30:
_Trac comment by cdknight on 2018-11-13 13:32:15:_

Two suggestions from the Joe at the CCB meeting: in SendMessageFull there are multiple buffers created in multiple if blocks. Move that up.

Also instead of passing a char* buffer into GetPipeName, change it to take a typedef struct that wraps the OSAL struct to remove the double-copy.

skliper, 2019-09-30:
_Trac comment by cdknight on 2018-11-14 15:40:07:_

Updated code at [changeset:c0a9d80]

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-02-25 16:16:53:_

Updates pending CCB review, expected test and merge disposition.

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-03-26 13:10:28:_

Note this may impact get pipe id by name #210

CDKnightNASA, 2019-11-08:
Something I'm thinking about...OS_GetQueueInfo copies the struct to a buffer, and GetPipeName will copy the name to a second buffer.

I'm sure the answer is NO but we trade efficiency for safety when we do all of this buffer copying. GetQueueInfo could just return the pointer to the memory of the queue's info and GetPipeName could return the pointer to the string from that queue info. Just thought as this is flight software, we could significantly reduce the memory stack use by not copying data around as much as we do.

jphickey, 2019-11-08:
I do agree this is a lot of unnecessary string copying action here.

I would be generally in favor of using the OS_queue_prop_t object directly.  But you'd have to instantiate that object at the parent function rather than a 20 character name buffer.  The overall stack usage would be less, though, and it would avoid unnecessary string copies.

CDKnightNASA, 2019-11-21:
In discussing GetPipeIdByName, the CCB agreed to make GetPipeName a public API. As there's interactions between GetPipeIdByName and GetPipeName, I merged the branch for this ticket into the branch for ticket #210 (pull req #404 ). I am closing this ticket and suggesting we review/approve/merge the other ticket/branch. I just pushed changes to support a public GetPipeName.

closed 2019-11-21 by CDKnightNASA

skliper, 2019-12-30:
Marking as duplicate (of #210) for ticket cleanup/management