nasa/cFE #295: CFE_ES_ScanAppTable() possible race conditions

skliper, 2019-09-30:
The `CFE_ES_ScanAppTable()` is called from the ES message processing thread, but it does not lock the global ES data structure when reading/writing from the global data.

ES software bus command messages are safe because this function is called by the same thread that is processing those messages, therefore concurrency is not possible here.

But other functions, like `CFE_ES_DeleteApp()` and `CFE_ES_ReloadApp()` are part of the public API and these also update the same fields within the app state data structure that `CFE_ES_ScanAppTable()` is reading.  `CFE_ES_RunLoop()` can also modify fields within this structure and this is called by pretty much every app in the system.

This issue was noted while reviewing the fix for a similar issue in #279.

skliper, 2019-09-30:
_Imported from trac issue 264.  Created by jphickey on 2019-03-25T17:01:55, last modified: 2019-06-24T11:44:20_

acudmore, 2020-03-27:
CFE_ES_RunLoop, CFE_ES_DeleteApp, and CFE_ES_ReloadApp all lock ES shared data already.
It may be possible just to guard the call to CFE_ES_ScanAppTable in the ES main task.
Before doing that, it is probably worth a more thorough analysis of how this data is locked in these various calls. We don't want to introduce a deadlock by just locking/unlocking data around the CFE_ES_ScanAppTable when ES calls it.

jphickey, 2020-03-29:
I've looked into this already and I have a proposal/plan which is mostly implemented but just needs to be tested on all the various platforms.

The proposal is to introduce a more consistent notion of a "background job" in ES.  We have several of these scattered about, including the Performance Log dump (separate ticket #456) and they are all implemented differently and inconsistently.

Worth noting that ScanAppTable has other issues besides the locking, in particular the timers are based on "scan intervals" which are not consistent, as they happen after every message (incl. SCH "send tlm") as well as after a period of message inactivity.

At any rate, my proposal is to have ES start a single, low-priority thread for all of these background jobs.  The jobs will be implemented as state machines which perform some (limited) work on each invocation.  The prototype will be:

`bool (*RunFunc)(uint32 ElapsedTime, void *Arg);`

The `ElapsedTime` parameter is the amount of time elapsed (based on OS clock) since the last time the job was executed, `Arg` is a generic/opaque state structure.  The function returns `false` if the job is idle (no work to do) or `true` if it is actively doing something.

Both the periodic ScanAppTable and the Performance Log file dump fit nicely into this model, as well as (probably) any other longer-running jobs that occur in ES.

PR #598 (2020-04-21): Fix #295, Resolve app table scanning race conditions
**Describe the contribution**

Fixes #295 
Fixes #480 

Create a new background job to handle the maintenance tasks that had been performed in the ES main task as part of the `CFE_ES_ScanAppTable()` routine.

All app state changes, including those invoked by messages, are now handled by this job.
    
This also slightly changes the semantics of CFE_ES_RunLoop and CFE_ES_ExitApp.  Now, the CFE_ES_RunLoop routine no longer requires a RunStatus buffer.  Instead, the only thing that matters is the RunStatus value that is eventually passed to CFE [... cut]
commit a4a48bd5 (2020-04-21): Merge pull request #598 from jphickey/fix-295-apptable-scan

Fix #295, Resolve app table scanning race conditions
commit bf249865 (2020-04-21): Fix #295, Move app table management to background task

Create a new background job to handle the maintenance tasks
that had been performed in the ES main task as part of the
CFE_ES_ScanAppTable() routine.

All app state changes, including those invoked by messages,
are now handled by this job.

This also slightly changes the semantics of CFE_ES_RunLoop and
CFE_ES_ExitApp.  Now, the CFE_ES_RunLoop routine no longer requires
a RunStatus buffer.  Instead, the only thing that matters is the
RunStatus value that is eventually passed to CFE_ES_ExitApp after
the shutdown is complete.  This should be [... cut]
closed 2020-04-27 by astrogeco