nasa/cFE #298: CFE_FS_Decompress infinite loop on truncated gzip file

skliper, 2019-09-30:
If CFS_FS_Decompress is given a truncated gzip file, it will enter an infinite loop in which it attempts to read more data from the file, gets nothing, and tries again. Discovered by accidentally attempting to have ES load a new compressed application (which we didn't know had been truncated), which led to ES getting stuck and an eventual watchdog reset.

The problem seems to be that running out of bytes in a gzipped file before decompression is finished is not considered an error. I can see how this might be intentional if it is expected that the file handle might be a stream that could present data after being emptied, but for the normal file use case I think it is a bug. Suggested fix by combining the two checks at line 309 of cfe_fs_decompress.c:

if ( State->insize == 0 ) return EOF;

if ( len == OS_FS_ERROR ) 
{
    State->Error = CFE_FS_GZIP_READ_ERROR;
    return EOF;
}

into one: 

if ((State->insize == 0) !|| (len == OS_FS_ERROR)) {
    State->Error = CFE_FS_GZIP_READ_ERROR;
    return EOF;
}

Recommend investigation if insize might temporarily hit 0 during a normal decompression.

Reported via email from Mike Stewart, mike@capellaspace.com

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-07-03 12:48:08:_

Moved unfinished 6.6.1 issues to next minor release

thewonderidiot, 2019-10-04:
Hi, Mike here! I can't send the file that originally caused this problem for us (EAR...) but I've made another one that causes the same issue. Starting with this image:
[image]

If I compress and the truncate the image as follows:
```
gzip corgi.jpg
dd if=corgi.jpg.gz of=corgi2.jpg.gz bs=2048 count=1
```
This gives me this file: [corgi2.jpg.gz](https://github.com/nasa/cFE/files/3692305/corgi2.jpg.gz)

If I uplink this to our flight computer and then try to extract it via FM_DECOMPRESS, ES_START_APP, etc., whatever app tries to do the decompression gets stuck in an infinite loop.

Let me know if this file is able to reproduce it for you!

closed 2019-10-23 by jphickey

skliper, 2020-05-12:
Won't fix, deprecated

commit b7e13070 (2019-10-23): Fix #298, fix inifinite loop