nasa/cFE #299: CFE_ES_ReloadApp() and CFE_ES_DeleteApp() missing range check on AppID

skliper, 2019-09-30:
These two functions are defined in `cfe_es.h` and therefore available for external applications and libraries to call.

These accept a passed in `AppID` parameter which is used as a table index, but they do not check the range of the value prior to using it as an array index.  If the value is out of range, this will cause undefined behavior and possibly a crash.

skliper, 2019-11-16:
Just piling on CFE_ES_SetAppState, and likely others (needs scrub per #300)

dmknutsen, 2020-03-09:
I finished the API scrub.  It is attached.
Also note that CFE_ES_SetAppState does not show up in the audit.  That is because it is an internal function.
[API_Audit.xlsx](https://github.com/nasa/cFE/files/4307811/API_Audit.xlsx)

skliper, 2020-04-08:
Duplicate of #546 (complete survey of issues).

closed 2020-04-08 by skliper