nasa/cFE #300: Scrub all public API calls to ensure that all input parameters are properly sanitized before use

skliper, 2019-09-30:
As a matter of policy, all CFE function calls that are available for use by external apps or libraries (i.e. all headers in the `fsw/cfe-core/src/inc` subdirectory) should sanity-check their input values before using them.

During recent code reviews it was noted that at least two of these functions were missing a required range check, as noted in bug #299.

This task is to perform an additional review on the public API calls and ensure that proper input value sanitization is being performed.

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-04-01 16:19:55:_

As part of this scrub also confirm failures are reported as errors, and unit testing adequately covers the failure cases.

Closing #273 as a duplicate of this effort (also somewhat a duplicate of #299).

skliper, 2019-09-30:
_Trac comment by jhageman on 2019-07-03 12:48:08:_

Moved unfinished 6.6.1 issues to next minor release

dmknutsen, 2020-03-09:
I finished the API scrub. It is attached.
Also note that CFE_ES_SetAppState does not show up in the audit. That is because it is an internal function.
[API_Audit_v2.xlsx](https://github.com/nasa/cFE/files/4308042/API_Audit_v2.xlsx)

skliper, 2020-03-16:
See #546 #547 #548 #549 

The API scrub effort was completed, resolution tickets written so no further action required on this ticket.

closed 2020-03-16 by skliper