nasa/cFE #302: Time Services need cleanup relative to requirements

skliper, 2019-09-30:
Per CCB on 3/27/19, Time Services goes way beyond it's requirements.  Need to re-evaluate configuration options and reduce mission specific code.

skliper, 2020-02-19:
Suggestions - Explicitly define TAT pulse and 1Hz pulse, the system should handle various configurations (any combination of 'virtual/internal' and external/API, synced or not)

TAT pulse source selection (SetSource) should be generalized to 'virtual' or passed to PSP, or get triggered via API
1Hz pulse source selection (new) should be generalized to 'virtual' or passed to PSP, or get triggered via API
MET source selection (new) should be generalized to 'virtual' or passed to PSP, or passed in via API
1Hz pulse when external shall flywheel after mission config elapsed time without a pulse (missing requirement)

Really TIME ifdefs need to be removed, and TIME should just do what's requested when requested.

Note #518 will fix some of the requirements but not all, so revisit after that update.

skliper, 2020-02-19:
Note set clock state is not exactly the same as cTIME2012, the external time mode since it also does VALID/INVALID.  Code could be simplified.

skliper, 2020-02-19:
See cTIME2013 - need to simplify configuration to adjust SCTF when requested (regardless of mode).

skliper, 2020-02-19:
SetSourceCmd - not well defined requirement or implementation.  Suggest updates per the 1Hz comment above (sort of tied 1Hz cycle to Tone but needs work).

skliper, 2020-05-05:
CFE_TIME_Locaal1HzISR related design also needs cleanup and implementation is not consistent with CFE_TIME_Tone1HzISR.  Wouldn't expect an "ISR" to be exposed in an API, Tone1HzISR has an ExternalTone API wrapper, etc.  See also #551 which only fixed the duplicate prototype, but bigger issues should be addressed as part of refactor.

skliper, 2021-05-11:
Also noted in code review, inconsistent parameter names used in the APIs (Time1/2, TimeA/B).  Improve consistency.

skliper, 2021-05-11:
Also noted in code review, consider merge of external events (CFE_TIME_ExternalMET, CFE_TIME_ExternalGPS, CFE_TIME_ExternalTime).  Although might be simpler to just include all three and support selection (internal/virtual/external MET, OS/GPS/Time (and relation to tone), etc) instead of `#ifdefs` all over.  They are APIs, you don't have to use them.

thnkslprpt, 2023-05-20:
> Per CCB on 3/27/19, Time Services goes way beyond it's requirements. Need to re-evaluate configuration options and reduce mission specific code.
> 
> As part of the cleanup, factor out duplicate code. Specifically referenced at code review in command handling.
> 
> * `CFE_TIME_SetTimeCmd`, `CFE_TIME_SetMETCmd`, `CFE_TIME_SetStcfCmd`, and `CFE_TIME_SetLeapSeconds` are all basically the same logic, etc.
> ...

This first point is addressed here (excl. `CFE_TIME_SetLeapSeconds()` which does not have enough common logic to justify being included):
 - https://github.com/nasa/cFE/pull/2340

thnkslprpt, 2023-05-20:
> Also noted in code review, inconsistent parameter names used in the APIs (Time1/2, TimeA/B). Improve consistency.

This is addressed here:
- https://github.com/nasa/cFE/pull/2342

thnkslprpt, 2025-06-24:
> * `CFE_TIME_ToneSendGPS` and `CFE_TIME_ToneSendTime` are basically the same logic, refactor

The above is addressed here:
- https://github.com/nasa/cFE/pull/2658