nasa/cFE #313: Memory alignment issues in Software Bus

skliper, 2019-09-30:
The CFE code has some problem areas for CPUs that have strict memory alignment requirements (e.g. SPARC).  Casting from a type with smaller requirements to a type with larger requirements generates a compile time warning and may induce a runtime exception if the memory actually is not aligned.

The following errors exist in the SB subsystem:

{{{
/cfe/fsw/cfe-core/src/sb/cfe_sb_api.c: In function 'CFE_SB_ZeroCopyGetPtr':
/cfe/fsw/cfe-core/src/sb/cfe_sb_api.c:1853:12: warning: cast increases required alignment of target type [-Wcast-align]
     return (CFE_SB_Msg_t *)address;
            ^
/cfe/fsw/cfe-core/src/sb/cfe_sb_api.c: In function 'CFE_SB_ZeroCopyReleasePtr':
/cfe/fsw/cfe-core/src/sb/cfe_sb_api.c:1896:35: warning: cast increases required alignment of target type [-Wcast-align]
                                   (uint32 *) (((uint8 *)Ptr2Release) - sizeof(CFE_SB_BufferD_t)));
                                   ^
/cfe/fsw/cfe-core/src/sb/cfe_sb_buf.c: In function 'CFE_SB_GetBufferFromCaller':
/cfe/fsw/cfe-core/src/sb/cfe_sb_buf.c:107:30: warning: cast increases required alignment of target type [-Wcast-align]
    CFE_SB_BufferD_t    *bd = (CFE_SB_BufferD_t *)(((uint8 *)Address) - sizeof(CFE_SB_BufferD_t));
                              ^
/cfe/fsw/cfe-core/src/sb/cfe_sb_util.c: In function 'CFE_SB_GetMsgTime':
/cfe/fsw/cfe-core/src/sb/cfe_sb_util.c:280:21: warning: cast increases required alignment of target type [-Wcast-align]
         TlmHdrPtr = (CFE_SB_TlmHdr_t *)MsgPtr;
                     ^
/cfe/fsw/cfe-core/src/sb/cfe_sb_util.c: In function 'CFE_SB_SetMsgTime':
/cfe/fsw/cfe-core/src/sb/cfe_sb_util.c:345:21: warning: cast increases required alignment of target type [-Wcast-align]
         TlmHdrPtr = (CFE_SB_TlmHdr_t *) MsgPtr;
                     ^
}}}

NOTE: This is related to PSP ticket [cfs_psp:41]

skliper, 2019-09-30:
_Imported from trac issue 282.  Created by jphickey on 2019-04-25T09:54:14, last modified: 2019-08-14T14:09:53_

CDKnightNASA, 2020-01-03:
One option I'm considering, aligning the pointer and then declaring it aligned: https://gcc.gnu.org/onlinedocs/gcc-3.3/gcc/Type-Attributes.html

jphickey, 2020-01-03:
I thought I had a fix for at least some of these items..... let me check my un-merged branches.

In general I prefer to avoid anything compiler-specific unless absolutely necessary.

CDKnightNASA, 2020-01-03:
Ok, thanks; I am curious how to fix it without a compiler directive. 😊

jphickey, 2020-01-03:
My bad, I mistakenly wrote a new issue about this, not realizing that this one was already open - which is #437.

This doesn't fix everything though - we still have (genuine) issues with respect to casting the message buffer types - `CFE_SB_Msg_t` has larger alignment requirements than many of the messages themselves.

CDKnightNASA, 2020-01-29:
Seeing similar with ES:

```
/home/pi/cFS/cfe/fsw/cfe-core/src/es/cfe_es_shell.c: In function ‘CFE_ES_ShellOutputCommand’:
/home/pi/cFS/cfe/fsw/cfe-core/src/es/cfe_es_shell.c:178:41: error: cast increases required alignment of target type [-Werror=cast-align]
                     CFE_SB_TimeStampMsg((CFE_SB_Msg_t *) &CFE_ES_TaskData.ShellPacket);
                                         ^
/home/pi/cFS/cfe/fsw/cfe-core/src/es/cfe_es_shell.c:179:36: error: cast increases required alignment of target type [-Werror=cast-align]
                     CFE_SB_SendMsg((CFE_SB_Msg_t *) &CFE_ES_TaskData.ShellPacket);
                                    ^
/home/pi/cFS/cfe/fsw/cfe-core/src/es/cfe_es_shell.c:208:37: error: cast increases required alignment of target type [-Werror=cast-align]
                 CFE_SB_TimeStampMsg((CFE_SB_Msg_t *) &CFE_ES_TaskData.ShellPacket);
                                     ^
/home/pi/cFS/cfe/fsw/cfe-core/src/es/cfe_es_shell.c:209:32: error: cast increases required alignment of target type [-Werror=cast-align]
                 CFE_SB_SendMsg((CFE_SB_Msg_t *) &CFE_ES_TaskData.ShellP
```

CDKnightNASA, 2020-01-29:
similar with SB:

```
/home/pi/cFS/cfe/fsw/cfe-core/src/sb/cfe_sb_api.c: In function ‘CFE_SB_SubscribeFull’:
/home/pi/cFS/cfe/fsw/cfe-core/src/sb/cfe_sb_api.c:859:29: error: cast increases required alignment of target type [-Werror=cast-align]
       Stat = CFE_SB_SendMsg((CFE_SB_Msg_t *)&CFE_SB.SubRprtMsg);
                             ^
```

and TIME:

```
/home/pi/cFS/cfe/fsw/cfe-core/src/time/cfe_time_tone.c: In function ‘CFE_TIME_Tone1HzTask’:
/home/pi/cFS/cfe/fsw/cfe-core/src/time/cfe_time_tone.c:1220:28: error: cast increases required alignment of target type [-Werror=cast-align]
             CFE_SB_SendMsg((CFE_SB_Msg_t *) &CFE_TIME_TaskData.ToneSignalCmd);
                            ^
/home/pi/cFS/cfe/fsw/cfe-core/src/time/cfe_time_tone.c:1228:28: error: cast increases required alignment of target type [-Werror=cast-align]
             CFE_SB_SendMsg((CFE_SB_Msg_t *) &CFE_TIME_TaskData.ToneSendCmd);
                            ^
/home/pi/cFS/cfe/fsw/cfe-core/src/time/cfe_time_tone.c: In function ‘CFE_TIME_Local1HzTask’:
/home/pi/cFS/cfe/fsw/cfe-core/src/time/cfe_time_tone.c:1440:28: error: cast increases required alignment of target type [-Werror=cast-align]
             CFE_SB_SendMsg((CFE_SB_Msg_t *) &CFE_TIME_TaskData.Local1HzCmd);
                            ^
```

and in cmdUtil:

```
/home/pi/cFS/tools/cFS-GroundSystem/Subsystems/cmdUtil/SendUdp.c: In function ‘SendUdp’:
/home/pi/cFS/tools/cFS-GroundSystem/Subsystems/cmdUtil/SendUdp.c:90:20: error: cast increases required alignment of target type [-Werror=cast-align]
         inet_ntoa(((struct sockaddr_in*)result->ai_addr)->sin_addr), port);
                    ^
```

skliper, 2020-05-04:
Targeting 7.0 to allow major tlm packet updates.  Errors don't show up in 32bit build for MCP750 with CCSDS Version 2 set so not critical to 6.8.

closed 2020-05-08 by astrogeco