nasa/cFE #314: Memory alignment issues in Table Services

skliper, 2019-09-30:
The CFE code has some problem areas for CPUs that have strict memory alignment requirements (e.g. SPARC).  Casting from a type with smaller requirements to a type with larger requirements generates a compile time warning and may induce a runtime exception if the memory actually is not aligned.

The following errors exist in the TBL subsystem:

{{{
/cfe/fsw/cfe-core/src/tbl/cfe_tbl_internal.c: In function 'CFE_TBL_RemoveAccessLink':
/cfe/fsw/cfe-core/src/tbl/cfe_tbl_internal.c:480:70: warning: cast increases required alignment of target type [-Wcast-align]
             Status = CFE_ES_PutPoolBuf(CFE_TBL_TaskData.Buf.PoolHdl, (uint32 *)RegRecPtr->Buffers[0].BufferPtr);
                                                                      ^
/cfe/fsw/cfe-core/src/tbl/cfe_tbl_internal.c:492:74: warning: cast increases required alignment of target type [-Wcast-align]
                 Status = CFE_ES_PutPoolBuf(CFE_TBL_TaskData.Buf.PoolHdl, (uint32 *)RegRecPtr->Buffers[1].BufferPtr);
                                                                          ^
}}}

NOTE: This is related to PSP ticket [cfs_psp:41]

skliper, 2020-05-04:
Targeting 7.0 to allow major tlm packet updates. Errors don't show up in 32bit build for MCP750 with CCSDS Version 2 set so not critical to 6.8.

jphickey, 2020-05-05:
This is already fixed in issue #437, via commit 4607e4ee

closed 2020-05-05 by jphickey

skliper, 2020-05-05:
Fixed by #479