nasa/cFE #325: Potential alignment errors when allocating cfs tables.

skliper, 2019-09-30:
The buffers allocated for tables force a 32 bit alignment when the over all memory buffer is 64 bit aligned.

If a 64 bit variable is the first data type an alignment error will be reported.

If the first two variables in the data structure BD_t in the file cfe_esmempool.h are allocated as 32 instead of 16 the alignment of the table will match the buffer.

The workaround is to rearrange the table items to fall on what ever alignment is necessary.

skliper, 2019-09-30:
_Imported from trac issue 294.  Created by dmills3 on 2019-05-22T14:01:06, last modified: 2019-05-23T16:43:54_

skliper, 2019-11-01:
Insufficient info for me to understand what the issue is.  Please provide links to code, explicit steps to recreate issue, etc.

skliper, 2019-11-01:
Note did test a table with 64 bit value as first entry and there were no errors.

jphickey, 2019-11-01:
I have two questions here:
- What version of CFE was this originally reported against? (some alignment logic changed as part of CFE 6.6)
- Was the address of the table actually 64-bit aligned or was the test simply that it didn't throw an exception?

Note that x86-64 is traditionally lax about misaligned values.  The system will detect the misalignment and transparently do a non-aligned read, which is slower but it will not generate an exception.  So x86-64 is not a good test platform for this.  I believe SPARC is much more strict about alignments.

We probably do have some alignment issues here and there still remaining in the system, though, so its worth a scrub.

skliper, 2019-11-01:
Copy that, trying to get in touch with dmills3 to get more information (platform, versions, etc).

dmills3, 2019-11-04:
> I have two questions here:
>
> * What version of CFE was this originally reported against? (some alignment logic changed as part of CFE 6.6)
> * Was the address of the table actually 64-bit aligned or [...]

dmills3, 2019-11-04:
The CFE version was 6.5.0.  

The problem was noticed on the Leon3 which is a SPARC processor. I agree that the X86 family will not cause an exception but the extra resource used to realign are wasted when all that is needed is to force 64 bit alignment.

To recreate the issue all you have to do is create a table that has a double as the first variable and run with a SPARC processor.

The CheckBits and Allocated variables were changed from uint16 to uint32.

typedef struct
{
  uint32    CheckBits;
  uint32    Allocated;
  uint32    Size;
  uint32   *Next;
} OS_PACK BD_t;

jphickey, 2019-11-04:
@dmills3  is it possible you can test with CFE 6.6.0 (or even better, the current github master branch) on your SPARC hardware?

The CFE 6.6.0 release already includes a number of changes to ensure that the outputs from `CFE_ES_PoolCreateEx()` are better aligned for the target processor, and I strongly suspect your issue is already resolved if you update.

I do not have access to any SPARC hardware for local testing.  Not sure about GSFC.

dmills3, 2019-11-06:
I will try CFE 6.6.0

dmills3, 2020-01-15:
I changed something in the kernel and I am now unable to reproduce the issue. Not sure what I changed.

skliper, 2020-01-15:
I'll close for now (not reproducible), but if you encounter the error in the future we can reassess.  Thanks for checking.

closed 2020-01-15 by skliper

jphickey, 2020-01-15:
Note that CFE 6.5 does align to 4 byte boundaries so it basically has a 50/50 chance of also being 8-bit aligned depending on the previous allocations.  Any change to the effective order of operations or the base address of the pool itself could change its behavior.

So it does not surprise me that a non-aligned buffer issue in 6.5.0 might appear/disappear by seemingly unrelated changes.  Upgrading to at least the 6.6.0 release if possible should provide more stable behavior, as this was one of the known issues addressed in that release.