nasa/cFE #344: Remove fsw/mission_inc and fsw/platform_inc (classic build support)

skliper, 2019-09-30:
Directories are only used by classic build and duplicate defines in sample_defs.  Should be removed.

Targeted for 7.0 since it will break classic build (deprecated in 6.7).

skliper, 2019-11-01:
@jphickey @acudmore what are our remaining dependencies on these directories?  Could one of you list them here?

jphickey, 2019-11-01:
I don't believe CFE has any dependencies on these at all.   But sometimes apps directly include them (e.g. SAMPLE_APP or the LAB apps maybe?).  

Should be an obvious build failure if there was a dependency though.  My recommendation is to take these out, and grep for "mission_inc" or "platform_inc" globally across the code base to verify that nothing directly references it.

If those strings don't exist in any source files and the code builds and runs OK, can be pretty sure that nothing needed them.

PR #577 (2020-04-21): Fix #344, Remove mission/platform include dirs
**Describe the contribution**
Fixes #344 
Removes classic build directories.

**Testing performed**
Steps taken to test the contribution:
1. Tested w/ CI bundle branch on fork, see: https://travis-ci.com/github/skliper/cFS/builds/157335615

**Expected behavior changes**
None, reduces user confusion due to duplicate defines

**System(s) tested on**
 - Hardware: CI (AMD)
 - OS: Ubuntu 18.04
 - Versions: Master w/ this commit

**Additional context**
Only dependence found in framework was in cFS-GroundStation.  See XXX

**Third party code**
None

**Co [... cut]
commit 07c4caea (2020-04-21): Fix #344, Remove mission/platform include dirs
commit 0cd9f5e7 (2020-04-21): Merge pull request #577 from skliper/fix344-rm-mission-platform-inc

Fix #344, Remove mission/platform include dirs
closed 2020-04-27 by astrogeco