nasa/cFE #355: Add flag at prep stage to OMIT_DEPRECATED

skliper, 2019-09-30:
Need an easy way for users (and the CI system) to omit all deprecated elements.  Something like

make OMIT_DEPRECATED=true prep

which would then internally add all the -DCFE_OMIT_DEPRECATED -DOSAL_OMIT_DEPRECATED and so on flags when building everything.

skliper, 2019-11-01:
@jphickey do you have a suggestion for this one?  I quickly jammed in:
```
  # Omit deprecated elements if set
  if (OMIT_DEPRECATED)
    add_definitions(-DCFE_OMIT_DEPRECATED_6_6 -DOSAL_OMIT_DEPRECATED)
  endif (OMIT_DEPRECATED)
```
to the prepare function of mission_build.cmake so I could move forward with CI improvements locally.  Didn't cache but easy enough to add.

jphickey, 2019-11-01:
I recommend just setting OSAL_USER_C_FLAGS when doing the initial prep, which allows one to add your own flags.

i.e. `make SIMULATION=native OSAL_USER_C_FLAGS='-DCFE_OMIT_DEPRECATED_6_6 -DOSAL_OMIT_DEPRECATED -Wall -Werror -std=c99 -pedantic' prep`

Shouldn't need to edit files.  This is how I generally test it out.  This is also cached so it is retained as part of the build tree.

skliper, 2019-11-01:
I am really not a fan of doing it that way.  OSAL_USER_C_FLAGS is not consistently applied everywhere, I think I've either emailed you or commented somewhere about previous issues with this technique.  I'd much rather from a top level add these definitions explicitly, and not require the user or CI to know what the actual define values are.

skliper, 2019-11-04:
See #24 for some other issues with OSAL_USER_CFLAGS.  Not a fan of `make SIMULATION=native OSAL_USER_C_FLAGS='-DCFE_OMIT_DEPRECATED_6_6 -DOSAL_OMIT_DEPRECATED -Wall -Werror -std=c99 -pedantic' prep` since a layer of abstraction (like "OMIT_DEPRECATED") means I wouldn't have to retrain users whenever a flag changes or we decide to add or remove from the warning set.  Also allows us to work towards compliance (define an initial set, and add as needed) with only requiring to update in one spot... we've discussed strict-protoypes and cast-align in the past.

jphickey, 2019-11-05:
Yes, the top-level wrapper makefile is another place that we could put logic like this.  However it was never _required_ that projects use this wrapper, it is just a convenient way to build with a single "make" and not be concerned with the specific CMake invocation to generate your makefiles.

That being said, we probably do need to have some sort of document (or at least a paragraph in the README) that indicates the name and meaning of the various configure-time options in the build system for that version of code.

skliper, 2019-11-07:
CCB 20191106 - discussed as high priority so simple CI improvements can move forward (matrix jobs to build both with and without deprecated elements) without hard coding defines in yml scripts

PR #463 (2020-01-21): Add global scope option to omit deprecated items
**Describe the contribution**

Fix #355

Adds a "global_build_options.cmake" file akin to the existing arch_build/mission_build option files.  Include an example of this file that optionally does add_definitions() to omit the deprected elements for build testing.

**Testing performed**
Build with and without the OMIT_DEPRECATED flag on the prep command, and confirm that it is correctly translated to the `add_definition()` globally for all tools and FSW code.

**Expected behavior changes**
Affects build system only.  No change to runtime behavior [... cut]
commit 80ecc3ec (2020-01-21): Merge pull request #463 from jphickey/jph-fix-335-omit-deprecated

Fix #355, Add global scope option to omit deprecated items
commit b5c369a9 (2020-01-21): Fix #355: Add global scope option to omit deprecated items

Adds a "global_build_options.cmake" file akin to the existing
arch_build/mission_build option files.  Include an example of
this file that optionally does add_definitions() to omit the
deprected elements for build testing.
PR #490 (2020-01-27): Integration Candidate 20200121
**Describe the contribution**
Fix #459, fix #355, fix, #24, fix #101, fix #412, fix #189

**Testing performed**
Steps taken to test the contribution:
1. Current CI passes
1. Enhanced CI in #40 passes debug/with deprecated elements testing

**Expected behavior changes**
- Resolve doxygen warnings
- add OMIT_DEPRECATED flag
- Add strict warnings and enforce
- Always increment sequence counter
- Fix comments
- Add code coverage test to evs

**System(s) tested on**
 - CI: Ubuntu 18.04

**Additional context**
N/A

**Third party code**
N/A

**Contributor Info - All inf [... cut]
closed 2020-01-27 by skliper