nasa/cFE #358: User/beginner friendly permissive mode implementation

skliper, 2019-09-30:
Failure when running as a normal user due to not setting the permissive mode should be user friendly.  Both for message queues and priorities failures.

Also requesting a command line option, or default to permissive mode for the bundle (should just work out of the box on a linux system for a new user, the bundle is not a flight distribution).  Maybe relate it to SIMULATION=native setting?  Note current way it's set up (as an #undef in default_osconfig.h precludes passing it in as an option). 

Worth discussing since currently the bundle defaults to debug mode with no optimization anyways.  Perhaps permissive mode is also appropriate?  Distribution guide (or whatever documentation is available) could cover transition to more flight-like configuration.

skliper, 2019-11-01:
@jphickey @acudmore what's your preferred solution?  Right now the #undef requires a file edit to make it work which is undesirable from a CI point of view.  How about just change to a commented out #define, and add a prep flag like ENABLE_PERMISSIVE (and mention the prep flag in the default_osconfig.h)?

jphickey, 2019-11-01:
In the future we had discussed moving all "code selection" options in osconfig.h to a build system config rather than putting these in a C header file.  If/When we do that then the ENABLE_PERMISSIVE build option would logically be part of that.

In the meantime, I think we might be able to buy time by changing the `#undef` to be:

```
#ifdef SIMULATION
#define OSAL_DEBUG_PERMISSIVE_MODE
#endif
```

This assumes that new users build with `SIMULATION=native` though.  Should make sure that this is in the user guide.  By default I think the users guide has one do a "cross" build (non-sim) but using the host compiler so the output is for the same architecture in the end.

skliper, 2019-11-04:
I'm actually not a fan of mixing SIMULATION and the PERMISSIVE mode (contrary to my original comments on this ticket) since they really are unique as you have highlighed.  How about just take the setting out of osconfig.h, and do it in the build system instead of adding an intermediate step?  It can still just be a -D flag passed in, but get's me the easier configuration capability now.

If this approach is accepted, should also remove setting from:
osal/src/bsp/pc-linux/config/osconfig.h

See c19536c

skliper, 2019-11-06:
CCB - suggest just commenting out a #define, then allows environment variable to set
- suggest evaluating on how often something changes
- Suggest code selection is in build system
  - Things like network code, etc
- Suggest build options in build system (BUILDTYPE)
   - permissive mode fits under this option, suggestion to keep it unique

skliper, 2020-01-21:
CCB 20200115 - Discussed just adding native_osconfig.h and override the permissive mode setting, that way whenever running as SIMULATION=native the permissive mode is set.

PR #505 (2020-03-02): Fix #358, Add native permissive mode configuration
**Describe the contribution**
This adds the native_osconfig.h which is included after
the default_osconfig.h if SIMULATION=native is set.
Fix #358

**Testing performed**
Steps taken to test the contribution:
1. copy cfe/cmake/sample_defs to top dir (per standard setup instructions)
1. make SIMULATION=native prep
1. make; make install;
1. Execute core-cpu1 in build/exe/cpu1 and confirmed permissive mode
1. Also did steps above with just make prep, and confirmed permissive mode wasn't set

**Expected behavior changes**
No longer requires sed "ha [... cut]
commit ea00d467 (2020-03-02): Fix #358, Add native permissive mode configuration

This adds the native_osconfig.h which is included after
the default_osconfig.h if SIMULATION=native is set.
commit eea202c5 (2020-03-02): Merge pull request #505 from skliper/fix358-native-permissive-mode

Fix #358, Add native permissive mode configuration
PR #544 (2020-03-12): IC 20200304
**Describe the contribution**
Integration Candidate 20200304

Fixes #358, #487, and #498 

See PRs #505, #504, and #500 for more info
 
**Testing performed**
Steps taken to test the contribution:
1. Build steps '...'
1. Execution steps '...'

**Expected behavior changes**

#358 -  No longer requires sed "hack" to change the setting in default_config.h

#487 - Clean build

#498 - None

**System(s) tested on**
CI

**Additional context**
See PRs #505, #504, and #500 for more info

**Contributor Info - All information REQUIRED for consideration of pull request**
Jacob Hageman - NASA/GS [... cut]
closed 2020-03-12 by astrogeco