nasa/cFE #45: CFE_TIME_GetTime() should not return a structure

skliper, 2019-09-30:
In general it is not a good idea to return a structure from a function because compilers do not all perform this the same way.  Some compilers/ABIs are reasonably efficient at this (as gcc seems to be), but others are not so efficient and will do extra copies of the structure.

skliper, 2020-06-03:
I suggest we fix this for 7.0.  Also problems in SB that I'm addressing.

skliper, 2020-08-31:
Need to resolve how to handle this.  Deprecate the old and and add one with a new name?  Or breaking change as part of 7.0 and just fix it to be consistent?

jphickey, 2020-08-31:
I had assumed this would be address as part of a more comprehensive CFE TIME cleanup, as there are so many other issues with this subsystem (#78, #109, #123, and probably more?).

With respect to this particular API we should probably first discuss whether there is still a need/justification for the `CFE_MISSION_TIME_CFG_DEFAULT_TAI`/`CFE_MISSION_TIME_CFG_DEFAULT_UTC` platform config options.  It is another one of those things that is never really tested - the default has TAI so that is likely the only thing verified.  It also seems like apps that depend on the notion of civil/calendar time would need to know whether they are getting UTC or TAI so they'd be better off calling CFE_TIME_GetTAI or CFE_TIME_GetUTC directly.  I am not sure what the CFE_TIME_GetTime() is supposed to be for, really.

Perhaps add `CFE_TIME_GetCurrentTAI()` and `CFE_TIME_GetCurrentUTC()` that fill the specified output buffer and return `int32`, and deprecate all current time getter APIs that return a structure directly?

CDKnightNASA, 2020-08-31:
Hmm, seems like an opportune time to make cFE Time a modular structure as with cFE msg?

skliper, 2020-08-31:
Good points.  Not a critical change for 7.0, makes sense to me to include as part of the comprehensive cleanup (and possible modularization) post 7.0.

astrogeco, 2020-09-02:
[CCB 2020-09-02](https://github.com/nasa/cFS/wiki/Community-CCB-Agenda:-September-02-2020) Will include in bigger time services update

skliper, 2021-04-20:
Additionally analyze all CFE_TIME_SysTime_t pass by value cases and resolve.  Used in a handful of APIs:
```
cfe_evs.h:CFE_Status_t CFE_EVS_SendTimedEvent(CFE_TIME_SysTime_t Time, uint16 EventID, uint16 EventType, const char *Spec, ...)
cfe_fs.h:CFE_Status_t CFE_FS_SetTimestamp(osal_id_t FileDes, CFE_TIME_SysTime_t NewTimestamp);
cfe_msg.h:CFE_Status_t CFE_MSG_SetMsgTime(CFE_MSG_Message_t *MsgPtr, CFE_TIME_SysTime_t Time);
cfe_time.h:CFE_TIME_SysTime_t CFE_TIME_GetTime(void);
cfe_time.h:CFE_TIME_SysTime_t CFE_TIME_GetTAI(void);
cfe_time.h:CFE_TIME_SysTime_t CFE_TIME_GetUTC(void);
cfe_time.h:CFE_TIME_SysTime_t CFE_TIME_GetMET(void);
cfe_time.h:CFE_TIME_SysTime_t CFE_TIME_GetSTCF(void);
cfe_time.h:CFE_TIME_SysTime_t CFE_TIME_Add(CFE_TIME_SysTime_t Time1, CFE_TIME_SysTime_t Time2);
cfe_time.h:CFE_TIME_SysTime_t CFE_TIME_Subtract(CFE_TIME_SysTime_t Time1, CFE_TIME_SysTime_t Time2);
cfe_time.h:CFE_TIME_Compare_t CFE_TIME_Compare(CFE_TIME_SysTime_t TimeA, CFE_TIME_SysTime_t TimeB);
cfe_time.h:CFE_TIME_SysTime_t CFE_TIME_MET2SCTime(CFE_TIME_SysTime_t METTime);
cfe_time.h:void CFE_TIME_ExternalMET(CFE_TIME_SysTime_t NewMET);
cfe_time.h:void CFE_TIME_ExternalGPS(CFE_TIME_SysTime_t NewTime, int16 NewLeaps);
cfe_time.h:void CFE_TIME_ExternalTime(CFE_TIME_SysTime_t NewTime);
cfe_time.h:void CFE_TIME_Print(char *PrintBuffer, CFE_TIME_SysTime_t TimeToPrint);
```