nasa/cFE #372: cFE core services can create files FM can't delete

thewonderidiot, 2019-10-10:
**Describe the bug**
A few commands to the cFE core services, and possibly some cFS applications, are able to create files that cannot then be moved, renamed, deleted, or otherwise by FM. One example is `EVS_WRITELOG2FILE` (`CFE_EVS_FILE_WRITE_LOG_DATA_CC`). This command (and all others that take filenames in the cFE core) don't check the validity of a filename; they simply pull the filename from the command and call `OS_creat()` or `OS_open()` immediately.

FM, on the other hand, calls `CFS_IsValidFilename()` to check for validity of provided filenames in all commands. So, if a command like `EVS_WRITELOG2FILE` is used to create a file with an invalid filename, FM cannot then access that file in any way.

**To Reproduce**
Steps to reproduce the behavior:
1. Use `EVS_WRITELOG2FILE` to create a filename in `/ram` with a `?` in its filename.
2. Attempt to delete this file with FM.

**Expected behavior**
I would expect that no cFE application, or cFE itself, should be able to create files with invalid filenames.

My suggested fix, and how we've fixed this in our local copies of cFE and OSAL, is to move the `CFS_IsValidFilename()` function into the OSAL, and have all OSAL file API functions perform the check before operating on files. This guarantees that nothing above the OSAL can create a file with an invalid name.

**Code snips**
* [EVS creation of log data files without validating filename](https://github.com/nasa/cFE/blob/6aaf5a3244e02af43c8ebb893ef413916dff5ef2/fsw/cfe-core/src/evs/cfe_evs_log.c#L178)
* [FM check on valid filenames](https://github.com/nasa/FM/blob/a1ebbc8b48a90764837cd083fae6d00f8967d46f/fsw/src/fm_cmd_utils.c#L159)
* [cfs_lib filename validation](https://github.com/nasa/cfs_lib/blob/871ce1e8cc1e597464923029e04f2ff3aa47e64e/fsw/src/cfs_utils.c#L170)

**System observed on:**
 - Capella flight computer
 - OS: FreeRTOS 9.0.0
 - Versions: cFE 6.5.0, OSAL 4.2.1 (plus in-house FreeRTOS port), cfs_lib 2.2.0, FM 2.5.2

**Additional context**
This is another one we discovered accidentally operationally. One of our operators created an EVS log with a question mark in the name, and then we realized that we couldn't do anything with that file with FM. We ended up patching the filename check in cfs_lib with MM so it would allow for us to delete the file.

**Reporter Info**
Mike Stewart, Capella Space.

jphickey, 2019-10-11:
I do concur that all filename validation should be done in OSAL.  The CFS apps/library should not introduce a more restrictive filename pattern.   This is a problem with FM/CFS lib, not CFE though.

However I don't think OSAL API needs to be extended.  OSAL already does checks as part of any file operation.  If any OSAL call that accepts a filename is passed an invalid file name, that call should fail.   I'm not sure what value a separate validation call would add, because I don't think the app response/handling would be any different.

I would recommend to move this bug against FM.  The group managing the CFS apps should consider deprecating/removing this extra validation.  Just pass the filename to OSAL and let OSAL reject it if it is not valid.

thewonderidiot, 2019-10-11:
That works too! I wasn't sure where to file this one since there's a lot of pieces involved. :smile:  Should I close this issue and open a new one on FM?