nasa/cFE #374: Task ID Not Valid in CFE_ES_DeleteApp

skliper, 2019-10-10:
**Describe the bug**
When deleting an app the Task ID Not Valid message was observed:
```
1980-012-14:06:24.50399 ES Startup: TST_ES4 loaded and created                                                                                             
EVS Port1 66/1/CFE_ES 6: Started TST_ES4 from /ram/tst_es4.o, AppID = 8                                                                                    
EVS Port1 66/1/TST_ES4 1: TST_ES4: Version 2.0.0.0: Task Initialized                                                                                       
1980-012-14:07:01.50000 CFE_ES_DeleteApp: Delete Application TST_ES4 Initiated                                                                             
1980-012-14:07:06.00100 CFE_ES_GetTaskInfo: Task ID Not Valid: 8                                                                                           
EVS Port1 66/1/CFE_ES 8: Stop Application TST_ES4 Completed.
```
The app was still deleted.

**To Reproduce**
Steps to reproduce the behavior:
1. Should be able to recreate by just deleting an app (say sample_app) on VxWorks6.9 OSAL
(I haven't had a chance to verify myself)

**Expected behavior**
AppID = 8, TaskID seem to start in the 655xx range.  I'd expect the code to use the right values in the right calls.

**Code snips**
Needs investigation

**System observed on:**
 - mcp750
 - OS: VxWorks6.9
 - Versions: rc-6.7 bundle

**Additional context**
Seen during build verification testing.

**Reporter Info**
Jacob Hageman - NASA/GSFC
Found by Walt M.

jphickey, 2019-10-11:
Is it possible to get a stack trace when the error occurs?  Or can you compile your same mission config on Linux via e.g. SIMULATION=native and reproduce the error, where it is possible to get a stack trace?

Of note here, is that that `CFE_ES_DeleteApp()` does not seem to invoke `CFE_ES_GetTaskInfo()`, at least not that I can see.  A stack trace would show whether this is really from the delete call, or if it happened in a different call that happened just after it.

However, I do see that the `CFE_ES_ListTasks()` function does invoke `CFE_ES_GetTaskInfo()` and also invokes it improperly.  It is calling it with an array index, whereas it should be calling it with a task ID.

Basically, the line:
`Result = CFE_ES_GetTaskInfo(&TaskInfo,i);`
should probably be:
`Result = CFE_ES_GetTaskInfo(&TaskInfo,CFE_ES_Global.TaskTable[i].TaskID);`
instead.

I will push a fix for the above, but I'd still like to explain why you saw this in relation to `CFE_ES_DeleteApp` somehow.

skliper, 2019-10-13:
Looks like the commit fixed the events in the delete app, is more information still needed?

jphickey, 2019-10-15:
Nope, its all fixed.  I was able to reproduce it locally.  This should be good to go.

commit 198abc6b (2019-10-31): Fix #374: CFE_ES_GetTaskInfo must be passed a task ID

Fix 2 areas where the CFE_ES_GetTaskInfo() function could
have been passed an App ID rather than a task ID, which
would create an incorrect result.
PR #388 (2019-11-07): Integration Candidate 20191030
**Describe the contribution**
Fixes #361, fixes #373, fixes #374, fixes #381

**Testing performed**
Steps taken to test the contribution:
1. Checked out bundle with OSAL and cFE ic-20191030 branches 
1. make ENABLE_UNIT_TESTS=TRUE SIMULATION=native prep
1. make
1. make install
1. make test
Built without warnings, all tests passed except osal_timer_UT (nominal result on linux)
executed cfe, successful startup with no warnings

**Expected behavior changes**
Resolved potential lockup bug
Resolved anomalous messages produced during app delete

**System(s) tested on:* [... cut]
closed 2019-11-07 by skliper