nasa/cFE #383: doxygen warning: explicit link

avan989, 2019-10-30:
**Describe the bug**
doxygen gives warning: explicit link request to 'CFE_SB_CmdHdr_t' could not be resolved ...etc

[userguide_warning_explicit_link.txt](https://github.com/nasa/cFE/files/3789214/userguide_warning_explicit_link.txt)

**To Reproduce**
Steps to reproduce the behavior:
1. make usersguide

**System observed on:**
 - Hardware
 - Ubuntu 18.04
 - doxygen 1.8.13, rc-6.7-0

**Reporter Info**
Anh Van, NASA Goddard

avan989, 2019-11-04:
Explicit link warning is caused by cross-referencing *.h files not being include in doxygen build. 

For Example: CFE_SB_BufferD_t is being cross referenced by cfe/cmake/sample_defs/cpu1_platform_cfg.h but CFE_SB_BufferD_t is being defined src/sb/cfe_sb_priv.h which is not included in the doxygen build.

In addition some of the warning are given by struct being defined in several different PSP/OSAL. For example: CFE_PSP_ExceptionHook is being defined for linux, rtems, and mcp750-vxworks. 

Possible Solution:

1)  I get rid of all cross reference to get rid of these warning.

2)  I include all the files in the build. But for this I would need to also implement logic in mission_build.cmake to choose the correct files based on the what psp/osal users choose.

jphickey, 2019-11-04:
I think the solution choice is dependent on the context and probably needs to be evaluated on a case by case basis.

For instance the `CFE_SB_BufferD_t` is a private, CFE-internal data structure and not exposed in the public API.  Therefore I would think this should also _not_ be referenced in the "user guide" style API documentation.

However, if generating a CFE developers guide, then I would think all the internal files should be indexed in this case.

The problem of a developer guide is that the specifics are different for each CPU within a multi-CPU mission.  So to be correct one would have to generate a _separate_ dev guide for each of the CPUs within the mission.  AFAIK the compromise here was just to build a single dev guide but potentially include more than one OSAL/PSP in it, and that is probably what triggers the warnings.

skliper, 2019-11-07:
CCB 20191107 - discussed, with the forward path of just using the first target's configuration (OSAL and PSP) for default linkage where required.  This is what CI will use (so linux/pc by default).  Projects can always doxygen with a different configuration if they really want to, but by picking a default the warnings can be resolved.

PR #400 (2019-11-15): Fix #383 - Resolve Explicit Link 
**Describe the contribution**
Fix #383, resolving explicit link and any remaining misc. warning. 

**Testing performed**
Steps taken to test the contribution:
1. make usersguide
2. Verify warning is gone

**System(s) tested on:**
 - Hardware
 - Ubuntu 18.04
 - rc-6.7-0

**Contributor Info**
Anh Van, NASA Goddard
commit 7142938e (2019-11-15): Fix #383, resolve explicit link and any remaining misc.
commit 9aedaaba (2019-11-30): 20191106/13: Merge #387 #390 #391 #393 #395 #400

Fixes #379, Fixes #380, Fixes #383, Fixes #384,
Fixes #385, Fixes #392
Code reviewed and approved at 20191106 and 20191113 CB
PR #413 (2019-11-30): Integration Candidate 20191115
**Describe the contribution**
Fixes #379, Fixes #380, Fixes #383, Fixes #384,
Fixes #385, Fixes #392

**Testing performed**
Steps taken to test the contribution:
1. Checked out bundle with ic-20191115 branches 
1. make ENABLE_UNIT_TESTS=TRUE SIMULATION=native prep
1. make
1. make install
1. make test
Built without warnings, all tests passed except osal_timer_UT (nominal result on linux)
executed cfe, successful startup with no warnings

Confirmed passed CI (with cppcheck enforced!)

**Expected behavior changes**
No functional changes.
- doxygen/cppcheck warning/e [... cut]
closed 2019-11-30 by skliper