nasa/cFE #389: Doxygen configuration cleanup

skliper, 2019-11-01:
**Is your feature request related to a problem? Please describe.**
1. The EXTRACT_ALL flag set to YES defeats the WARN_IF_UNDOCUMENTED and WARN_NO_PARAMDOC enabled errors
1. Warning files should be unique to each doxygen created doc (currently set in the cfe-common.doxyfile.in) to allow parallel runs and unique responses in CI

**Describe the solution you'd like**
Set EXTRACT_ALL to NO (likely will need a follow-on ticket to resolve all the new warnings)
Define WARN_LOGFILE in each top doxyfile.in

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2021-01-06:
Note the warning file updates will require updating CI that checks those files.

astrogeco, 2021-01-06:
> Note the warning file updates will require updating CI that checks those files.

Which update?

skliper, 2021-01-06:
> > Note the warning file updates will require updating CI that checks those files.
> 
> Which update?

If the warning file name is updated to be unique per document build (suggestion in 2. above).