nasa/cFE #394: coding style guide

CDKnightNASA, 2019-11-05:
**Is your feature request related to a problem? Please describe.**
Code and file naming in cFS and apps is inconsistent and does not follow a particular set of guidance, and no guidance were developed previously for cFS.

**Describe the solution you'd like**
Develop a "Style Guide" in markdown format for inclusion in cFE/docs, describing code formatting conventions, symbol naming conventions, documentation conventions, directory structure and file naming conventions, etc. This should be referenced by the App Developer's guide and any other relevant docs and may incorporate or replace parts of existing documents. Perhaps also link/refer from the github wiki.

**Describe alternatives you've considered**
Other options include adding style information to the app developer's guide or identifier naming conventions documents.

**Additional context**
Jake provided the GSFC "indent" flags that are standard at GSFC, this is a good starting point for formatting and should be documented.

**Requester Info**
Chris Knight, NASA Ames Research Center

CDKnightNASA, 2019-11-05:
Should this be a ticket in the cFS github site and should docs be moved "up" into the cFS repository?

CDKnightNASA, 2019-11-05:
Joe: In particular, I'd like to see a mention of NOT using type suffixes (u/l) on literals. Been "bitten" by this numerous times.  (as in #335)

skliper, 2019-11-05:
I suggest suffixes on literals also get documented in the deployment/configuration guide.

jphickey, 2019-11-05:
I think all documentation should remain with CFE.  The "cFS" repo is just a test bundle; it shouldn't provide anything that cannot be gotten via the submodules.

CDKnightNASA, 2019-11-06:
clang-format seems like a better option, as a number of editors (vim, emacs, bbedit, ...) can be integrated with the configuration file used by clang-format.