nasa/cFE #398: cfe_psp_start timer race condition, leading to undef memory writes

semussan, 2019-11-06:
Hi I'm working with cFE in the context of NASA's icarus codebase

We think we found an issue, In psp/fsw/pc-linux/src/cfe_psp_start.c, a timer is setup with this line:
    setitimer (ITIMER_REAL, &timer, NULL);
The timer event ultimately calls CFE_ES_PerfLogAdd after 250ms which starts making assignments on the perf object like so:
    Perf->MetaData.InvalidMarkerReported = TRUE;
 
The problem is there is a time race condition in that CFE_ES_MAIN_FUNCTION’s call to CFE_ES_SetupPerfVariables has not yet initialized the Perf object, depending on your execution speed. This is unlikely but if you are adding initialization code after setitimer it will likely lead to undefined behavior
 

**To Reproduce**
Steps to reproduce the behavior:
after 
   setitimer (ITIMER_REAL, &timer, NULL);
add
   sleep(30);

And undefined behavior will happen (or nothing observable). It was crashing our lengthy DLL initialization that we were running in the place of sleep(30);

**Expected behavior**
No Crash

**Code snips**
N/A

**System observed on:**
 - 64 bit intel
 - OS: vmware workstation 14 running Ubuntu 18 vm, host is windows 10
 - Versions current

**Additional context**
N/A

**Reporter Info**
Reachable at tyler.young@jhuapl.edu. This isn't currently an issue for us but is probably worth reporting

jphickey, 2019-11-06:
This issue is already fixed in the current master branch.  The 1Hz callback is invoked by CFE_TIME itself, which happens only after CFE_ES has completed its full initialization.

skliper, 2019-11-06:
@semussan Could you confirm the PSP version number you saw this on?