nasa/cFE #402: Missing cFE Deployment Guide

mfeliu, 2019-11-12:
**Describe the bug**
The `cFE Deployment Guide` named in the `cFE Application Developers Guide.doc` is not included in the repository and is not easily to find.

**Reporter Info**
Marco A. Feliu - *National Institute of Aerospace*

skliper, 2019-11-12:
Unfortunately the diversity (and associated complexity) of deployments throughout the user community have exceeded the original scope of the cFE Development Guide.  The most useful and up-to-date open source documentation is delivered as part of the repository (see README.md in cmake for building, docs directory for general documentation, generated doxygen guides, or the source code).  If there are specific "open source" questions I suggest reaching out either to the community email list (see https://lists.nasa.gov/mailman/listinfo/cfs-community) or posting an issue.  For direct project support for any major work I suggest pursuing a support agreement through the related organizations.

mfeliu, 2019-11-12:
Thank you. I thought someone had forgotten to include the old `cFE Deployment Guide` in the up-to-date repository. If the reason for not including it is that is considered outdated, this issue can be closed. Thanks again for your help.

skliper, 2019-11-12:
Good note that we should update the references in the `cFE Application Developers Guide.doc`, thanks!

closed 2019-11-12 by skliper