nasa/cFE #409: Clarify CFE_ES_SYSLOG_APPEND use vs CFE_ES_WriteToSysLog

skliper, 2019-11-17:
**Is your feature request related to a problem? Please describe.**
CFE_ES_SYSLOG_APPEND is used is just a few places (cds, esmempool), everywhere else is CFE_ES_WriteToSysLog.  

**Describe the solution you'd like**
Consistency is preferred for maintenance and usability,  If there isn't sufficient justification for a separate macro, remove it and use CFE_S_WriteToSysLog. 

**Describe alternatives you've considered**
If there is clear justification, consider adding to the macro documentation/comments

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2019-11-17:
Looks like it's used when CFE_ES_SysLog_snprintf has already been called on the passed in string, outside/after unlocking.  I guess this gives slightly more accurate time stamp, vs just building the non-timestamped string and passing it to WriteToSysLog.

jphickey, 2019-11-18:
The two API functions serve a different purpose.  The "Append" function writes a pre-formatted string directly to the log, whereas the "WriteToSyslog" uses a printf-style formatting scheme to create the message, which is subsequently appended to the log.  

The latter makes another copy of the message on the stack (post-formatting) whereas the former does not.  Also, the ES lock needs to be held for the appending part, whereas it does not for the formatting part.

One would have to significantly change e.g. `CFE_ES_CDSBlockWrite` to use `CFE_ES_WriteToSyslog` due to the locking differences.  The function determines the outcome (and any log messages) while holding the lock for the CDS mem pool. It then releases that lock and does the append, which takes the ES lock.  It is done this way to avoid holding both at the same time.

Ideally, it would be nice to defer the formatting of the log message too, as there is no reason to hold the CDS pool lock while formatting the error message.  Then `CFE_ES_WriteToSyslog` could be used here too.  But as this only happens in off-nominal cases it isn't really worth the optimization.

skliper, 2019-11-18:
> One would have to significantly change e.g. `CFE_ES_CDSBlockWrite` to use `CFE_ES_WriteToSyslog` due to the locking differences. The function determines the outcome (and any log messages) while [...]
Disagree, no significant change needed (in fact you describe the easy alternative below).  Just build the base LogMessage within the lock, let CFE_ES_WriteToSyslog add the timestamp at the end.

> Ideally, it would be nice to defer the formatting of the log message too, as there is no reason to hold the CDS pool lock while formatting the error message. Then `CFE_ES_WriteToSyslog` could be used here too. But as this only happens in off-nominal cases it isn't really worth the optimization.
Exactly.  I'm fine with no change though.

jphickey, 2019-11-18:
But then you would essentially be double-buffering your message strings on the stack, doubling stack use and making it less efficient for no real benefit other than wanting to call the same API.  Both APIs would still to exist regardless, so may as well use the right tool for the right job.

My recommendation is to simply extend the documentation on these, if the current documentation isn't clear enough.

skliper, 2019-11-18:
I understand CFE_ES_WriteToSyslog creates a copy on the stack.  It's apparently ok everywhere else CFE_ES_WriteToSyslog is used, I'm not clear why this additional stack use isn't ok here in the interest of consistency.  Also would remove the query to TIME services from within the lock.  I bring it up since it wasn't obvious to me why a different implementation was necessary for the three cases where the CFE_ES_SYSLOG_APPEND macro is used.  Is there a unique stack use concern for these three cases?

I don't consider consistency in the context of maintainability "no real benefit".

jphickey, 2019-11-18:
All the other places where `CFE_ES_WriteToSyslog` is invoked do not have a pre-formatted message already.  They are calling the function with the intent of doing both, formatting and appending.  So in the `WriteToSyslog` use cases only one string buffer will be instantiated on the stack.  If there is a case where this function is used with a preformatted message, it would be better to use Append instead (but I don't see one).

Can you clarify:
>Also would remove the query to TIME services from within the lock

As far as I can tell, the call to TIME services is not inside a lock - at least not the ES lock - as it is done as part of the message formatting (snprintf), not appending.  

Specifically in the CDS case it will be done while holding the CDS mempool lock because the CDS code calls the `CFE_ES_SysLog_snprintf` function while holding this.  It's not ideal but not terrible either, because the CFE_TIME implementation does gets the reference time in a lockless, time-limited manner, so it will not deadlock.  As I said, it would be nice to fix this, but it is not a major issue by any measure.

Either way I think it is correct API design that the Syslog routines differentiate between _formatting_ of a message (which requires no special synchronization) and _appending_ of that message into the log area (which does require synchronization).   This is why the separate "snprintf" and "append" actions exist.  The `WriteToSyslog` is merely a convenience function that does both, hence why it is used in a majority of cases.

skliper, 2019-11-18:
> All the other places where `CFE_ES_WriteToSyslog` is invoked do not have a pre-formatted message already. They are calling the function with the intent of doing both, formatting and appending. So in [...]

Perfect.  Got it.

> Can you clarify:
>
>
> As far as I can tell, the call to TIME services is not inside a lock - at least not the ES lock - as it is done as part of the message formatting (snprintf), not appending. [...]

I should have been more specific, the CDS mempool lock is what I was referring to.

> Either way I think it is correct API design that the Syslog routines differentiate between _formatting_ of a message (which requires no special synchronization) and _appending_ of that message into [...]

Agreed, API design is fine.  Probably would have been more clear to me if WriteToSyslog called snprintf and append.

jphickey, 2019-11-18:
Upon additional review - the only "extra" thing that `CFE_ES_WriteToSyslog` does is a final call to `OS_printf` to duplicate the syslog message onto the console, which wouldn't be done when calling the functions individually.  

In particular, this means that any syslog messages generated in the CDS use case wouldn't appear on the console.  This could be a valid concern/issue that might warrant a fix.

Also not that the console write itself is a preformatted message, and subject to the same double-formatting/double-buffering concerns.  It would be nice to pass it directly to `OS_ConsoleWrite` rather than `OS_printf` to avoid this.  The former exists and is implemented as if it were a public API, but unfortunately the prototype is missing from osapi-os-core.h so right now it cannot be directly invoked by apps.

closed 2020-01-02 by skliper