nasa/cFE #410: Initialize system log mode implementation doesn't match requirement cES1522

skliper, 2019-11-18:
**Is your feature request related to a problem? Please describe.**
Requirement cES1522 "Upon a Processor Reset, the cFE shall set the System Log Mode to discard."
Implementation:
https://github.com/nasa/cFE/blob/5d7d766941369b6e92a0328217e43d5fd02a8cf1/fsw/cfe-core/src/es/cfe_es_task.c#L236-L239

The build verification test just happens to pass because the sample configuration defaults the mode to discard.

**Describe the solution you'd like**
Update requirement or implementation.

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

jphickey, 2019-11-18:
Recommend the requirement be reworded to say "Upon a Processor Reset, the cFE shall set the System Log Mode to the platform default"

skliper, 2019-11-18:
I should have included the rational: "Want to preserve the System Events that may have captured the cause of the processor reset."  So intent was to specifically override on a processor reset.  Definitely open for discussion...

jphickey, 2019-11-18:
And one can achieve (or fail to achieve) the goal stated in the rationale in the rationale either way.  It is just as easy to lose the syslog information with it set to discard, as the log may have been full already when the event occurred.  At the end of the day this is just a system integration issue.  Seems pretty cut and dry to me - cfe cannot "enforce" that deployments are correctly configured.

skliper, 2019-11-18:
It's definitely one for the architects @acudmore @jwilmot.  I'd think the vision in mind was for typical use default to overwrite for power on reset, force to discard on processor reset.  User setting to discard at power on is certainly their choice to make, for which the discard on processor reset has no effect.

skliper, 2020-02-10:
Path forward - separate configurable default for power on and processor reset.  Default power on to overwrite and processor reset to discard.  Requirements will be updated to match (will like req issue to this one).  Users can modify however desired, but comes out of the box with the configuration that matches initial intent.

skliper, 2020-02-11:
See cES1522 and cES1523 updates in #509

PR #625 (2020-04-28): Fix #410, separate SYSLOG configurable defaults for power on and proc…
Describe the contribution
Fix #410 .
Separates configurable defaults for power on and processor reset. Default power on set to overwrite and default processor reset to discard.
Added an additional unit test to cover the path added to CFE_ES_TaskInit.

Testing performed:
Ran unit tests

Expected behavior changes
Upon power on reset, default system log mode set to overwrite.
Upon processor reset, default system log mode set to discard.

System(s) tested on
Oracle VM VirtualBox
OS: ubuntu-19.10
Versions: 6.7.12.0, OSAL 5.0.11. [... cut]
commit 2890418d (2020-04-28): Merge pull request #625 from dmknutsen/issue_410

Fix #410, separate SYSLOG configurable defaults for power on and proc…
commit 95a3a98f (2020-04-28): Fix #410, separate SYSLOG configurable defaults for power on and processor reset
closed 2020-05-01 by astrogeco