nasa/cFE #414: Suggested upper limit value creates infinite loop

impala454, 2019-11-19:
When the user sets `CFE_PLATFORM_SB_HIGHEST_VALID_MSGID`, the documentation specifies a range of 1 to 0xFFFF.  This value is later used in a `#define` in cfe_sb_priv.h as follows:

     #define CFE_SB_MAX_NUMBER_OF_MSG_KEYS   (1+CFE_PLATFORM_SB_HIGHEST_VALID_MSGID)

This value is then used as the upper limit of a for loop [here](https://github.com/nasa/cFE/blob/5d7d766941369b6e92a0328217e43d5fd02a8cf1/fsw/cfe-core/src/sb/cfe_sb_init.c#L224).
   
The problem is that if the user sets the upper limit of 0xFFFF as specified, the `CFE_SB_MAX_NUMBER_OF_MSG_KEYS` becomes `65536`.  This value is used as the upper range of the for loop linked above.  Since the iterator used in that loop is a unsigned 16 bit integer, it can never reach `65536` and creates an infinite loop.

Suggested fix would be to:   
1.  Stop using `#define`s which add to other `#define`s
2.  Never use `#define`s as iterator range variables
3.  Check types of all iterators
4.  Change the documentation of `CFE_PLATFORM_SB_HIGHEST_VALID_MSGID` to say max value of `0xFFFE`

skliper, 2019-11-19:
Suggestion: Resolve 4 as immediate change, split 1-3 into a more general code improvements ticket for `#define` use/handling.

Did you run static analysis on this configuration?  I suspect it would have also found this issue.

impala454, 2019-11-19:
I did not run any static analysis.  I suspect it would light up like a Christmas tree :)

skliper, 2020-06-05:
Closed by #708

PR #708 (2020-05-20): Fix #707, Resolve highest MsgID of 0xFFFF bug
**Describe the contribution**
Changes Message Key from uint16 to uint32 to avoid rollover and system hang
Fix #707 
Fix #414

**Testing performed**
Steps taken to test the contribution:
1. Set CFE_PLATFORM_SB_HIGHEST_VALID_MSGID to 0xFFFF
1. Built (SIMULATION=native) and ran, confirmed startup
1. CI - https://travis-ci.com/github/skliper/cFS/builds/166553342

**Expected behavior changes**
Full message ID range available

**System(s) tested on**
 - Hardware: cFS Dev 3
 - OS: Ubuntu 18.04
 - Versions: Bundle w/ this change

**Additional context**
Ide [... cut]
closed 2020-06-05 by skliper