nasa/cFE #418: many unit tests report event count errors as hex

CDKnightNASA, 2019-11-22:
**Describe the bug**
Unit tests check the number of events sent during the test, many of these in sb_UT.c are reported as hex...Such as in Test_SB_Cmds_SendPrevSubs (line 2704.)

**To Reproduce**
Uncovered when I was changing GetPipeName to produce events (as part of making it a public API) and the unit test event counts had to be updated. A simple test is to change the "ExpRtn" for a test.

**Expected behavior**
Event counts should reported in decimal.

**Code snips**
See above.

**System observed on:**
Linux VM

**Additional context**
N/A

**Reporter Info**
Christopher.D.Knight@nasa.gov

CDKnightNASA, 2019-11-22:
In reviewing the error reporting in general, I'm wondering if we want to use some macros and some global variables rather than having so much duplicated code in each fn.

jphickey, 2019-11-22:
Yes, I am a big fan of macro-izing many of the repetitive unit test constructs.  In particular there should be a at least macro to call a function and check its int32 return code (very common).   I have an example of such a macro in nasa/sample_lib#11

But updating all the existing test cases is extremely tedious and time consuming....

CDKnightNASA, 2019-11-22:
> But updating all the existing test cases is extremely tedious and time consuming....

*Raises hand to volunteer to do at least sb_UT.c as a "demo"*

jphickey, 2019-11-22:
> _Raises hand to volunteer to do at least sb_UT.c as a "demo"_
Great!!  No complaints from me about that!

CDKnightNASA, 2019-11-25:
Point of interest: my sample of sb_UT.c went from 10,487 lines of code to 6,878 lines, a roughly 35% reduction in code size for the UT. Check it out! :D

CDKnightNASA, 2020-01-22:
@skliper would like to review the sb_UT.c changes at an upcoming CCB meeting.

skliper, 2020-01-22:
@CDKnightNASA could you write an issue (Macro-ize unit test enhancement) and go ahead and put your changes in as a pull request to facilitate review?

CDKnightNASA, 2020-01-24:
> @CDKnightNASA could you write an issue (Macro-ize unit test enhancement) and go ahead and put your changes in as a pull request to facilitate review?

see https://github.com/nasa/cFE/pull/492

skliper, 2021-01-06:
Fixed in #698

closed 2021-01-06 by skliper