nasa/cFE #420: CFE_SB_DeletePipeFull event uses AppId instead of TaskId

CDKnightNASA, 2019-11-29:
**Describe the bug**
Near the end of CFE_SB_DeletePipeFull, it generates an event on successful deletion of the pipe. As part of the message, it's supposed to report the "owner" but it calls CFE_SB_GetAppTskName(Owner,FullName) where Owner is the AppId, not a (valid) TaskId. This causes CFE_SB_GetTaskInfo() to generate a syslog error and return an error code.

**To Reproduce**
Delete a pipe.

**Expected behavior**
Should produce a proper event. Unsure if it should report the Task name or the App name. Also should see if unit test should have caught this bug.

**Code snips**
cfe_sb_api.c#393  - CFE_EVS_SendEventWithAppID(CFE_SB_PIPE_DELETED_EID,CFE_EVS_EventType_DEBUG,CFE_SB.AppId,"Pipe Deleted:id %d,owner %s",(int)PipeId, CFE_SB_GetAppTskName(Owner,FullName));

**System observed on:**
Debian VM

**Additional context**
Add any other context about the problem here.

**Reporter Info**
Christopher.D.Knight@nasa.gov

closed 2020-01-07 by CDKnightNASA

skliper, 2020-01-07:
@CDKnightNASA did you mean to close this issue?  As far as I know it's still a bug.

CDKnightNASA, 2020-01-07:
commit 198abc6bfc631b589f57fa3f25e2b4711cd800d2 already addressed this in master.

skliper, 2020-01-07:
Duplicate of #374