nasa/cFE #424: Restart app delete's app prior to missing file check (requirement failure)

skliper, 2019-12-05:
**Describe the bug**
Requirement cES1007.2 - If the original cFE Application file is not found then the cFE shall reject the Command, increment the invalid Command counter, and generate an event message.
Rationale: Can't restart the Application if the original file has been removed. In this case, the Application will continue without a restart.

If you send a restart now with the file remove, the app exits prior to the restart failing.

**To Reproduce**
Steps to reproduce the behavior:
1. Normal startup (with sample_app)
1. Remove sample_app.so
1. Send ES Restart app for "SAMPLE_APP"
1. Observe error:
```
1980-012-14:05:46.50026 CFE_ES_RestartApp: Restart Application SAMPLE_APP Initiated
1980-012-14:05:47.00007 CFE_ES_ExitApp: Called with invalid status (1).
1980-012-14:05:47.00008 Application SAMPLE_APP called CFE_ES_ExitApp
EVS Port1 42/1/CFE_ES 14: Exit Application SAMPLE_APP on Error Completed.
```

**Expected behavior**
Behavior matches description, likely due an update to move behavior in rationale to requirement.  Should just check for file existing prior to delete of app.

**Code snips**
https://github.com/nasa/cFE/blob/3e60d950370423968744c198c4f10cd2d2252750/fsw/cfe-core/src/es/cfe_es_api.c#L250-L252

https://github.com/nasa/cFE/blob/3e60d950370423968744c198c4f10cd2d2252750/fsw/cfe-core/src/es/cfe_es_apps.c#L1079-L1095

**System observed on:**
 - cFS Dev server
 - OS: Ubuntu 16.04
 - Versions: current dev cFE 6.7.3

**Additional context**
None.

**Reporter Info**
Jacob Hageman - NASA/GSFC

jphickey, 2019-12-05:
Does this really need to be a requirement?  Seems like a superfluous test to me.

I recall this being discussed (or at least something similar) years ago, and this doesn't seem like a valid requirement.  This is a "restart", so the app was loaded once.  The only way it could not exist now would be if it was deleted between the time the app was first started and now, possibly due to a failed upload or something.  In other words, some type of operator intervention would have had to break it first.

But that is just one obvious failure mode.   Even with a "file exists" check there are plenty of other issues that could prevent this from working, which are not as predictable.  Maybe the file got corrupted, so it exists but it is un-loadable.  Maybe there isn't enough memory to load the new app, or the old one couldn't be unloaded because there was some dependency still bound to it.

It boils down to operator error.  The system can't prevent you from breaking it, nor should it try to be smarter than the ground system operators.  If someone sends a bad command, so be it.

skliper, 2019-12-06:
I have no objection to making the requirement match the code (remove the "Application will continue" part of the rational).  @acudmore or @jwilmot ?

skliper, 2019-12-06:
I'd basically add something like the following in the rationale "The current app is removed prior to attempt to load, so any issues with the restart (missing file, corrupted file, out of memory, etc) will leave the system with the app removed."

skliper, 2020-02-11:
See cES1007.2 updates in #509

skliper, 2020-02-11:
Requirement update done as part of #509, closing as duplicate.

PR #510 (2020-02-25): Fix #509,  Executive Services Requirements Scrub
**Describe the contribution**
Updates ES requirements, see details in issue.
Fixes #509
Closed #424 as a duplicate (fixed here).

**Testing performed**
None, only doc change

**Expected behavior changes**
Requirements now match code except where addressed by open issues (#484, #410, #399, #291)

**System(s) tested on**
N/A

**Additional context**
Note this commit is on top of #377 commits, just review this commit.

**Third party code**
N/A

**Contributor Info - All information REQUIRED for consideration of pull request**
Jacob Hageman - NASA/GSF [... cut]
closed 2020-02-11 by skliper