nasa/cFE #431: lgtm issue

avan989, 2019-12-17:
**Describe the bug**

cfe_sb_api.c
```
/* send an event for each pipe write error that may have occurred */
--
1489 | for(i=0;i < SBSndErr.EvtsToSnd; i++)
  | Comparison between i of type uint16 and EvtsToSnd of wider type uint32.
1490 | {
1491 | if(SBSndErr.EvtBuf[i].EventId == CFE_SB_MSGID_LIM_ERR_EID)
```

cfe_es_perf.c

```
if ((CmdPtr->TriggerMode >= CFE_ES_PERF_TRIGGER_START) && (CmdPtr->TriggerMode < CFE_ES_PERF_MAX_MODES))
--
  | Pointless comparison of unsigned value to zero.

```

cfe_es_perf.c

/* enable interrupts */
```
--
491 | OS_IntUnlock(IntFlags);
  | This expression has no effect (because OS_IntUnlock has no external side effects).
492 | }
493 | }

```

cfe_sb_task.c

```
if(CFE_PLATFORM_EVS_MAX_EVENT_FILTERS < CfgFileEventsToFilter){
--
  | Comparison is always false because 4 >= CfgFileEventsToFilter.

```

cfe_sb_api.c

```
/*
--
774 | * FIXME: If a hash or other conversion is used between MsgId and MsgKey,
775 | * then it is possible that this existing route is for a different MsgId.
776 | *
777 | * The MsgId should be checked against the "MsgId" in the route here.
778 | *
779 | * However it is not possible to have a mismatch in the default case where
780 | * MsgKey == MsgId
781 | */
  | FIXME comment: If a hash or other conversion is used between MsgId and MsgKey, [...]

```

**To Reproduce**
Steps to reproduce the behavior:
1. Build against lgtm

**Reporter Info**
Anh Van, NASA Goddard

skliper, 2020-02-13:
Note using clang the index >= 0 comes up as a warning, and will fail if built with -Werror.

skliper, 2020-03-23:
cfe_sb_api issue - CFE_PLATFORM_SB_MAX_DEST_PER_PKT description says:
`Destination number values that are checked against this configuration are defined by a 16 bit data word.`

The event buffer is only added to up to once per destination (Destinations is uint16 in CFE_SB_RouteEntry_t), so EvtsToSnd uint32 can be changed to uint16 to avoid this warning.

skliper, 2021-01-06:
OBE - new issues will be opened for any current LGTM warnings that require resolution.

closed 2021-01-06 by skliper