nasa/cFE #442: Recommended lgtm issues: Classes with too many fields

avan989, 2019-12-20:
**Is your feature request related to a problem? Please describe.**
lgtm issue classes with too many fields:

cfe_tbl_task.h
```
line 185 - 211
Struct struct <unnamed> has 24 fields; we suggest refactoring to 15 fields or fewer.

line 257 - 211
Struct struct <unnamed> has 16 fields; we suggest refactoring to 15 fields or fewer.

line 282:
Struct struct <unnamed> has 26 fields; we suggest refactoring to 15 fields or fewer - see 8 Values.
declaration of LastTblUpdated
group of 2 fields here
group of 2 fields here
group of 3 fields here
group of 3 fields here
group of 4 fields here
group of 5 fields here
group of 6 fields here
```

inc/cfe_sb_msg.h
```
line 543 - 578
Struct struct <unnamed> has 16 fields; we suggest refactoring to 15 fields or fewer.

line 613:
Struct struct <unnamed> has 16 fields; we suggest refactoring to 15 fields or fewer - see 2 Values.
group of 12 fields here
group of 4 fields here
```

inc/cfe_tbl_msg.h
```
line 705
Struct struct <unnamed> has 20 fields; we suggest refactoring to 15 fields or fewer - see 4 Values.
group of 2 fields here
group of 2 fields here
group of 8 fields here
group of 8 fields here

line 776 - 807
Struct struct <unnamed> has 17 fields; we suggest refactoring to 15 fields or fewer.
```

fs/cfe_fs_decompress.h
```
line 109 - 128
Struct struct <unnamed> has 16 fields; we suggest refactoring to 15 fields or fewer.
```

inc/cfe_es_msg.h
```
line 1487 - 1567
Struct struct <unnamed> has 38 fields; we suggest refactoring to 15 fields or fewer.
```

inc/cfe_es.h
```
line 207 - 250
Struct struct <unnamed> has 21 fields; we suggest refactoring to 15 fields or fewer.
```

inc/cfe_time_msg.h
```
line 993
Struct struct <unnamed> has 42 fields; we suggest refactoring to 15 fields or fewer - see 13 Values.
declaration of ClockStateFlags
declaration of DataStoreStatus
declaration of MaxLocalClock
declaration of VirtualMET
group of 10 fields here
group of 2 fields here
group of 2 fields here
group of 2 fields here
group of 2 fields here
group of 3 fields here
group of 5 fields here
group of 6 fields here
group of 6 fields here
```

sb/cfe_sb_priv.h
```
line 288 - 307
Struct struct <unnamed> has 19 fields; we suggest refactoring to 15 fields or fewer.
```

time/cfe_time_utils.h
```
line 170
Struct struct <unnamed> has 56 fields; we suggest refactoring to 15 fields or fewer - see 22 Values.
declaration of DataStoreStatus
declaration of Forced2Fly
declaration of Local1HzCmd
declaration of MaxDelta
declaration of MaxLocalClock
declaration of Spare
declaration of SynchCallback
declaration of ToneSendCmd
group of 15 fields here
group of 2 fields here
group of 2 fields here
group of 2 fields here
group of 2 fields here
group of 2 fields here
group of 2 fields here
group of 2 fields here
group of 2 fields here
group of 2 fields here
group of 2 fields here
group of 4 fields here
group of 4 fields here
group of 5 fields here
```

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context about the feature request here.

**Requester Info**
Full name and company/organization if applicable

skliper, 2019-12-22:
I suggest just globally turning this suggestion off (or suppressing).  Really just style, not a security issue.

avan989, 2019-12-23:
updated, reference https://github.com/nasa/cFS/pull/33 for fix.

closed 2019-12-23 by avan989