nasa/cFE #465: Incomplete coverage test for cfe_es_api.c

avan989, 2020-01-09:
**Is your feature request related to a problem? Please describe.**
current coverage: 97.7 %
Missing coverage for the following:

CFE_ES_SetAppState

```
     197             :         /* Caller error - invalid state */
     198           0 :         return;
```

CFE_ES_ReloadApp

```
     314           0 :            CFE_ES_SysLogWrite_Unsync ("CFE_ES_ReloadApp: Cannot Reload Application %s, File %s does not exist.\n",
     315           0 :                                   CFE_ES_Global.AppTable[AppID].StartParams.Name,
     316             :                                   AppFileName);
     317           0 :            ReturnCode = CFE_ES_FILE_IO_ERR;
```

CFE_ES_ExitApp

```
     471           0 :           CFE_ES_SysLogWrite_Unsync ("Application %s called CFE_ES_ExitApp\n",
     472           0 :                                  CFE_ES_Global.AppTable[AppID].StartParams.Name);
     473           0 :           CFE_ES_SetAppState(AppID, CFE_ES_AppState_STOPPED);
     474             : 
     475             :           /*
     476             :           ** Unlock the ES Shared data before suspending the app 
     477             :           */
     478           0 :           CFE_ES_UnlockSharedData(__func__,__LINE__);
     479             : 
     480             :           /*
     481             :           ** Suspend the Application until ES kills it.
     482             :           ** It might be better to have a way of suspending the app in the OS
     483             :           */
     484             :           while(1)
     485             :           {
     486           0 :              OS_TaskDelay(500);
```

CFE_ES_DeleteChildTask

```
    1272           0 :        CFE_ES_WriteToSysLog("CFE_ES_DeleteChildTask Error: Invalid Task ID: %u\n",(unsigned int)OSTaskId );
    1273           0 :        ReturnCode = CFE_ES_ERR_TASKID;
```

CFE_ES_GetAppIDInternal

`    1758           0 :       *AppIdPtr = 0;`

**Requester Info**
Anh Van, NASA Goddard

pepepr08, 2021-06-15:
`CFE_ES_SetAppState` no longer in cfe_es_api.c

pepepr08, 2021-06-15:
`CFE_ES_ReloadApp` tests are outdated. Comments don't match the actual test. I will be adding new tests for the whole function.

pepepr08, 2021-06-15:
Testing `CFE_ES_ExitApp` involves an infinite loop. This will be handled separately.

skliper, 2021-06-15:
> Testing `CFE_ES_ExitApp` involves an infinite loop. This will be handled separately.

Could you write a separate issue for any infinite loops you encounter?  These will be addressed separately.  This issue can be closed once the coverage possible with the current implementation is achieved.

pepepr08, 2021-06-15:
For `CFE_ES_DeleteChildTask`, the "invalid Invalid Task ID" error case was already being covered but code will be added to cover the "Task ID is not active" error case.

pepepr08, 2021-06-15:
`CFE_ES_GetAppIDInternal` has been removed, but UT code will be added to cover the null case on `CFE_ES_GetAppID`, `CFE_ES_GetAppIDByName` and `CFE_ES_RegisterCDS`

pepepr08, 2021-06-16:
> > Testing `CFE_ES_ExitApp` involves an infinite loop. This will be handled separately.
> 
> Could you write a separate issue for any infinite loops you encounter? These will be addressed separately. This issue can be closed once the coverage possible with the current implementation is achieved.

Created #1628 for infinite loop test case in `CFE_ES_ExitApp`

PR #1623 (2021-06-23): Fix #465, Added UT tests for cFE ES Api
**Describe the contribution**
Fixes #465 
Completed code coverage on the following functions:
- Test CFE_ES_ReloadApp bad AppID, core app, file doesn't exist
- Test CFE_ES_GetAppID: NULL AppID pointer
- Test CFE_ES_GetAppIDByName: NULL AppID pointer
- Test CFE_ES_GetAppIDByName: NULL name pointer
- Test CFE_ES_DeleteChildTask: Task inactive/invalid case
- Test CFE_ES_RegisterCDS: NULL handle and NULL name

**Testing performed**
Steps taken to test the contribution:
1.  Compiled code and tests for ES
1. Manually run coverage-es-ALL-testrunner and verify o [... cut]
commit 34f4b324 (2021-06-23): Merge pull request #1623 from pepepr08/fix465-es-api-ut

Fix #465, Added UT tests for cFE ES Api
commit a0e89a8a (2021-06-23): Fix #465, added UT tests for various ES API functions

Test CFE_ES_ReloadApp bad AppID, core app, file doesn't exist
Test CFE_ES_GetAppID: NULL AppID pointer
Test CFE_ES_GetAppIDByName: NULL AppID pointer
Test CFE_ES_GetAppIDByName: NULL name pointer
Test CFE_ES_DeleteChildTask: Task inactive/invalid case
Test CFE_ES_RegisterCDS: NULL handle and NULL name
PR #1632 (2021-06-23): cFE Integration candidate: 2021-06-22
## Context

Part of https://github.com/nasa/cfs/pull/270

## Testing

cFE Checks <https://github.com/nasa/cFE/pull/1632/checks>
Bundle Checks <https://github.com/nasa/cfs/pull/270/checks>

## Authors 

@skliper 
@zanzaben 
@jphickey 
@pepepr08 

## Description

### PR #1627

Fix #1626, rename/clean CFE coverage assert macros

### PR #1630 

Fix #1629, correct path to users guide warning log

### PR #1621 

Fix #1477, add additional test cases for Child Tasks

### PR #1608 
Fix #1367, Add cfe functional tests to CI

### PR #1623 

Fix #465, Added UT tests f [... cut]
closed 2021-06-23 by astrogeco