nasa/cFE #466: Incomplete coverage test for cfe_es_shell.c

avan989, 2020-01-09:
**Is your feature request related to a problem? Please describe.**

no coverage test for CFE_ES_ShellCountObjectCallback
```
    338           0 : static void CFE_ES_ShellCountObjectCallback(uint32 object_id, void *arg)
     339             : {
     340             :     uint32                 *CountState;
     341             :     uint32                 idtype;
     342             : 
     343           0 :     CountState = (uint32 *)arg;
     344           0 :     idtype = OS_IdentifyObject(object_id);
     345           0 :     if (idtype < OS_OBJECT_TYPE_USER)
     346             :     {
     347           0 :         ++CountState[idtype];
     348             :     }
     349           0 : }
```

**Requester Info**
Anh Van, NASA Goddard

skliper, 2020-02-25:
Looks like dead code to me, not called anywhere.  Recommend just delete since it's static.

skliper, 2020-02-25:
Also related to #484, plan to deprecate shell functionality from core cFE.

jphickey, 2020-02-26:
This is part of the `CFE_ES_ListResources` routine, referenced here:

https://github.com/nasa/cFE/blob/5802d229ce047185e9247192f0fa328219d54b78/fsw/cfe-core/src/es/cfe_es_shell.c#L370

Although this in turn is part of the "shell" ground command it is a separate function - if the "command" string supplied by the ground is a special string `ES_ListResources` then this routine is called and it is _NOT_ actually passed to the shell.

Not sure why this was overloaded like this and not simply made into a separate command (there are actually 3 of these "special" command strings).

jphickey, 2020-02-26:
Also note that this is precisely the use-case for the "wrappers" in other unit test code, to provide a way for UT to invoke a static function.  As the wrappers are not allowed the alternative is to remove the static and put the prototype into a private header that UT can get to.

skliper, 2020-02-26:
> Also note that this is precisely the use-case for the "wrappers" in other unit test code, to provide a way for UT to invoke a static function. As the wrappers are not allowed the alternative is to remove the static and put the prototype into a private header that UT can get to.

NONONONO, please no.  This is DEAD CODE.  We should not hack a work around to test dead code!

skliper, 2020-02-26:
Got it now, missed the call in ListResources.  Still I contend we should not be forcing unit tests on static functions, it's not externally accessible and shouldn't be directly called by external tests.  Static functions not expected or required to pass unit testing, since typically inputs are not sanitized.  Better design is to get coverage using the stub and tests that call exposed functions.

skliper, 2020-02-26:
Probably good to note I'm referring to "unit tests" as the GSFC standard that defines what a unit test has to do, and what a unit needs to pass.  These standard is not applicable to non-exposed elements since these don't fit the standards definition of a "unit".  The non-exposed code all still needs coverage, but the shell processing is flagged for removal anyways so not worth a redesign or test update #484.  Note again #including C files is a violation of the GSFC standards.  There are other acceptable approaches to get full code coverage.

jphickey, 2020-02-26:
Again, this is a matter of terminology.  "coverage tests" aka "white box" tests ARE supposed to exercise every internal line of code that executes, whether or not it is directly callable.  The reality is that during normal system operation this code __DOES__ execute.  Therefore it __DOES__ need to be verified in the path coverage analysis if that is worth anything.

For a black box functional test I concur that if it is not directly callable. it shouldn't be tested.  But that is __NOT__ the testing performed with mocks and stubs.  Mocks and stubs do not work for functional test.

jphickey, 2020-02-26:
Also note that this code here is _NOT_ part of the "Shell processing" that was discussed for removal.  This data is not part of a shell command and not generated by the shell.  It is an overloaded command where some special "commands" are not passed to a shell but handled locally.  This is one of those, and it is a separate problem from the deprecation/removal of shell commands.

skliper, 2020-03-02:
> Again, this is a matter of terminology. "coverage tests" aka "white box" tests ARE supposed to exercise every internal line of code that executes, whether or not it is directly callable. The reality [...]

Seems to me are talking past each other.  I said "non-exposed code still needs coverage" which agrees with your path coverage statement.  If you can get coverage during normal system ops without directly calling the static code, you can get coverage in a unit test (w/ stubs) without directly calling the static code.  Please refrain from continuing to push for a design approach that is in direct contradiction to our coding standards.

> Also note that this code here is NOT part of the "Shell processing" that was discussed for removal. This data is not part of a shell command and not generated by the shell. It is an overloaded [...]

We certainly should consider which of these special commands should become commands that follow the nominal command handling pattern, but this definitely needs requirements/code rework/refactor since it's currently not certifiable.

jphickey, 2020-03-02:
> If you can get coverage during normal system ops without directly calling the static code, you can get coverage in a unit test (w/ stubs) without directly calling the static code.

This is __not__ a true statement.  Mocks and stubs do not have the implementation of the full versions.   In particular when the implementation involves a callback, this is a problem.

> Please refrain from continuing to push for a design approach that is in direct contradiction to our coding standards.

How is this pushing a noncompliant design approach?  I thought I suggested removing the `static` qualifier and adding a prototype for it, which would be compliant, would it not? 

Either way though, the UT has to call the function somehow.  Either by a hook or directly.

skliper, 2020-03-02:
> This is **not** a true statement. Mocks and stubs do not have the implementation of the full versions. In particular when the implementation involves a callback, this is a problem.

The stub get's the callback pointer, why can't it be called?

>> Please refrain

Was related to...

> precisely the use-case for the "wrappers"

jphickey, 2020-03-02:
In my previous comment, 

> the alternative is to remove the static and put the prototype into a private header that UT can get to.

...is what I was suggesting here.

> The stub get's the callback pointer, why can't it be called?

Depends on the specifics of the code and test case.  In simple cases like this one it can probably be done using a hook, but in other cases where a registration action is done separately, it is not as easy to do.

Having UT invoke the internal function directly is also simpler (not requiring use of a hook function) and since it can handle all cases, not just the trivial ones, it should be the preferred approach.

skliper, 2020-03-02:
It is acceptable to remove static and call directly, I don't recall ever saying otherwise.

pepepr08, 2021-06-21:
There is no `cfe_es_shell.c` anymore.

pepepr08, 2021-06-23:
@astrogeco @skliper there is nothing to be done here. Can I close this ticket?

astrogeco, 2021-06-23:
The only thing I would check is if we want to "migrate it" to the apps team since we have a new shell app in the works. @ejtimmon thoughts?

skliper, 2021-06-28:
@astrogeco - the file name and function names won't be the same.  Better to just run coverage analysis on the new app vs try to adjust this issue (and I think @ejtimmon tracks tickets differently).

closed 2021-06-28 by skliper