nasa/cFE #473: Incomplete Unit test for src/time

avan989, 2020-01-09:
**Is your feature request related to a problem? Please describe.**
Current Coverage: 99.1 %

cfe_time_api.c
CFE_TIME_CFE2FSSeconds

```
     681           0 :         if (-ConvertFactor > SecondsCFE)
     682             :         {
     683           0 :             SecondsFS = 0;
     684             :         }
```

CFE_TIME_Print
```
     766           0 :         NumberOfMinutes++;
     767           0 :         NumberOfSeconds -= 60;
```

cfe_time_task.c
CFE_TIME_TaskInit
```
     352           0 :       CFE_ES_WriteToSysLog("TIME:Error subscribing to time at the tone request data cmds:RC=0x%08X\n",(unsigned int)Status);
     353           0 :       return Status;
```

cfe_time_utils.c
CFE_TIME_GetReference
```
  676           0 :         if (RetryCount == 0)
     677             :         {
     678             :             /* unsuccessful read */
     679           0 :             break;
     680             :         }
     681             : 
     682           0 :         --RetryCount;
     683             :     }
     684             : 
```

**Requester Info**
Anh Van, NASA Goddard

pepepr08, 2021-07-30:
In cfe_time_api.c, can't test the following lines when default CFE_MISSION_TIME_EPOCH_SECOND is 0.
```
        -:  621:    /*
        -:  622:    ** Convert the cFE time (offset from epoch) into calendar time...
        -:  623:    */
        4:  624:    NumberOfMinutes = (TimeToPrint.Seconds / 60) + CFE_MISSION_TIME_EPOCH_MINUTE;
        4:  625:    NumberOfSeconds = (TimeToPrint.Seconds % 60) + CFE_MISSION_TIME_EPOCH_SECOND;
        -:  626:
        -:  627:    /*
        -:  628:    ** Adding the epoch "seconds" after computing the minutes avoids
        -:  629:    **    overflow problems when the input time value (seconds) is
        -:  630:    **    at, or near, 0xFFFFFFFF...
        -:  631:    */
        8:  632:    while (NumberOfSeconds >= 60)
        -:  633:    {
    #####:  634:        NumberOfMinutes++;
    #####:  635:        NumberOfSeconds -= 60;
        -:  636:    }
```

skliper, 2021-08-11:
Note also need to call CFE_TIME_Local1HzISR per #548

pepepr08, 2021-08-13:
> In cfe_time_api.c, can't test the following lines when default CFE_MISSION_TIME_EPOCH_SECOND is 0.
>
> ```
>         -:  621:    /*
>         -:  622:    ** Convert the cFE time (offset from epoch) into [...]

Reached 100% in all but cfe_time_api.c

```
File '/home/jfmarti4/cfs-gateway/remote/cfs/cfe/modules/time/fsw/src/cfe_time_api.c'
Lines executed:98.96% of 192

File '/home/jfmarti4/cfs-gateway/remote/cfs/cfe/modules/time/fsw/src/cfe_time_task.c'
Lines executed:100.00% of 381

File '/home/jfmarti4/cfs-gateway/remote/cfs/cfe/modules/time/fsw/src/cfe_time_utils.c'
Lines executed:100.00% of 255

File '/home/jfmarti4/cfs-gateway/remote/cfs/cfe/modules/time/fsw/src/cfe_time_tone.c'
Lines executed:100.00% of 149
```

PR #1836 (2021-08-19): Fix #473 Adding coverage tests to cFE TIME
**Describe the contribution**
Fix #473 Completing coverage for cFE TIME. 

Completed coverage on all lines except two. See issue for details:
https://github.com/nasa/cFE/issues/473#issuecomment-898579608

Missing cov
**Testing performed**
Steps taken to test the contribution:
1. `make -C build/native/default_cpu1/time`
2. `make -C build/native/default_cpu1/time test`
3. `cd /home/jfmarti4/cfs-gateway/remote/cfs/build/native/default_cpu1/time/ut-coverage/CMakeFiles/coverage-time-ALL-object.dir/__/fsw/src`
4. `gcov -b gcov -b cfe_time_api.c.gcda`
5. `gc [... cut]
commit 54612f17 (2021-08-19): Merge pull request #1836 from pepepr08/fix473-time-coverage

Fix #473 Adding coverage tests to cFE TIME
commit eec8ec98 (2021-08-19): Fix #473 Adding coverage tests to cFE TIME
PR #1839 (2021-08-21): cFE Integration candidate: 2021-08-24
**Describe the contribution**

PR #1803 

Add software bus tests 

- Fix #1680 
- Fix #1681 
- Fix #1682

PR #1756 

- Fix #1755, separate variable for OSAL status 

PR #1809 

- Fix #1807, increase SB pool max size bucket 

PR #1842 

- Fix #1840, Add Null check for CFE_ResourceId_FindNext 

PR #1844 
- Fix #1840, Hotfix for strict resource IDs. 

PR #1828 

Improve tbl coverage tests 

- Fixes #472 
- Fixes #1792
- Fixes #1793
- Fixes #1795 
- Fixes #1791
- Fixes #1789
- Fixes #1788 

PR #1833 
- Fix #1824, Add missing cases for msg id func tests 

PR #1 [... cut]
closed 2021-08-21 by astrogeco