nasa/cFE #476: The difference in linking behavior results in build errors on macOS

stanislaw, 2020-01-11:
**Is your feature request related to a problem? Please describe.**

I am getting the following build error when building on macOS (more complete log below):

```
[ 24%] Linking C shared module sample_app.so
cd /sandbox/cFS/cmake-build-debug/cpu1/apps/sample_app && /Applications/CLion.app/Contents/bin/cmake/mac/bin/cmake -E cmake_link_script CMakeFiles/sample_app.dir/link.txt --verbose=1
/usr/bin/gcc -fPIC -Wall     -D_XOPEN_SOURCE=600    -g  -shared  -o sample_app.so CMakeFiles/sample_app.dir/fsw/src/sample_app.c.o CMakeFiles/sample_app.dir/fsw/src/sample_table.c.o 
Undefined symbols for architecture x86_64:
  "_CFE_ES_ExitApp", referenced from:
      _SAMPLE_AppMain in sample_app.c.o
  "_CFE_ES_PerfLogAdd", referenced from:
      _SAMPLE_AppMain in sample_app.c.o
```

Let's take a `_CFE_ES_ExitApp` symbol for a typical example: it is a symbol from `libcfe_core_default_cpu1.a` which does not exist yet at the moment when the `sample_table.c.o` above is built. This works on Linux because it seems to allow this kind of a forward declaration but it does not on macOS which fails immediately.

What macOS expects here is an explicit `-lcfe_core_default_cpu1` to be added to the compilation line above:

```
/usr/bin/gcc -fPIC -Wall     -D_XOPEN_SOURCE=600    -g  -shared  -o sample_app.so CMakeFiles/sample_app.dir/fsw/src/sample_app.c.o CMakeFiles/sample_app.dir/fsw/src/sample_table.c.o -lcfe_core_default_cpu1
```

I recognize that CFS/CFE uses a two-stage CMake generation process when one CMake tree calls to the second which builds the cpu1-related code and at this point, it looks like on Linux
the ` sample_app.so ` is compiled from the first tree and then with its dependencies forward-declared it gets linked to a final executable when built in the second tree and that's where its undefined symbols are resolved.

I have created a minimal example that demonstrates this difference in the linking behavior on Linux and macOS [on Github](https://github.com/stanislaw/Examples/tree/20200110-dynamic-static-forward-declaration-mac-vs-linux) and asked [a question on StackOverflow](https://stackoverflow.com/questions/59687933/linking-a-dynamic-library-that-links-in-symbols-from-a-static-library-macos-vs) to see if I can make macOS/Clang to match the behavior of GCC/Linux but so far I am not aware of what could be done to resolve this.

**Describe the solution you'd like**

This issue makes me think that the order of how things are built by CFS/CFE could be changed. macOS linking errors indicate that we are building something in the first CMake tree that depends on a cpu1-specific code that is built from the second CMake tree. I could see it practical to move the compilation and linkage of all cpu1-specific code to the second stage so that the order of dependencies is strictly direct and there are nowhere any forward-declared undefined symbols of this kind in the binaries.

**Describe alternatives you've considered**

- I will try to see if I can find any linker flags in `clang --help` that could activate the Linux-like linking behavior but I am not sure if such flags exist.

- I can create my custom set of CMakeLists.txt files to streamline the build process and eliminate the need in "forward declarations" but this will result in my own custom build system for CFS/CFE which I would rather like to avoid.

**Additional context**

Please check the StackOverflow question above.

**Requester Info**

Stanislav Pankevich

---

Full build error log:

```
[ 24%] Linking C shared module sample_app.so
cd /sandbox/cFS/cmake-build-debug/cpu1/apps/sample_app && /Applications/CLion.app/Contents/bin/cmake/mac/bin/cmake -E cmake_link_script CMakeFiles/sample_app.dir/link.txt --verbose=1
/usr/bin/gcc -fPIC -Wall     -D_XOPEN_SOURCE=600    -g  -shared  -o sample_app.so CMakeFiles/sample_app.dir/fsw/src/sample_app.c.o CMakeFiles/sample_app.dir/fsw/src/sample_table.c.o 
Undefined symbols for architecture x86_64:
  "_CFE_ES_ExitApp", referenced from:
      _SAMPLE_AppMain in sample_app.c.o
  "_CFE_ES_PerfLogAdd", referenced from:
      _SAMPLE_AppMain in sample_app.c.o
  "_CFE_ES_RegisterApp", referenced from:
      _SAMPLE_AppMain in sample_app.c.o
  "_CFE_ES_RunLoop", referenced from:
      _SAMPLE_AppMain in sample_app.c.o
  "_CFE_ES_WriteToSysLog", referenced from:
      _SAMPLE_AppInit in sample_app.c.o
      _SAMPLE_ProcessCC in sample_app.c.o
      _SAMPLE_GetCrc in sample_app.c.o
  "_CFE_EVS_Register", referenced from:
      _SAMPLE_AppInit in sample_app.c.o
  "_CFE_EVS_SendEvent", referenced from:
      _SAMPLE_AppMain in sample_app.c.o
      _SAMPLE_AppInit in sample_app.c.o
      _SAMPLE_ProcessCommandPacket in sample_app.c.o
      _SAMPLE_ProcessGroundCommand in sample_app.c.o
      _SAMPLE_VerifyCmdLength in sample_app.c.o
      _SAMPLE_NoopCmd in sample_app.c.o
      _SAMPLE_ResetCounters in sample_app.c.o
      ...
  "_CFE_SB_CreatePipe", referenced from:
      _SAMPLE_AppInit in sample_app.c.o
  "_CFE_SB_GetCmdCode", referenced from:
      _SAMPLE_ProcessGroundCommand in sample_app.c.o
      _SAMPLE_VerifyCmdLength in sample_app.c.o
  "_CFE_SB_GetMsgId", referenced from:
      _SAMPLE_ProcessCommandPacket in sample_app.c.o
      _SAMPLE_VerifyCmdLength in sample_app.c.o
  "_CFE_SB_GetTotalMsgLength", referenced from:
      _SAMPLE_VerifyCmdLength in sample_app.c.o
  "_CFE_SB_InitMsg", referenced from:
      _SAMPLE_AppInit in sample_app.c.o
  "_CFE_SB_RcvMsg", referenced from:
      _SAMPLE_AppMain in sample_app.c.o
  "_CFE_SB_SendMsg", referenced from:
      _SAMPLE_ReportHousekeeping in sample_app.c.o
  "_CFE_SB_Subscribe", referenced from:
      _SAMPLE_AppInit in sample_app.c.o
  "_CFE_SB_TimeStampMsg", referenced from:
      _SAMPLE_ReportHousekeeping in sample_app.c.o
  "_CFE_TBL_GetAddress", referenced from:
      _SAMPLE_ProcessCC in sample_app.c.o
  "_CFE_TBL_GetInfo", referenced from:
      _SAMPLE_GetCrc in sample_app.c.o
  "_CFE_TBL_Load", referenced from:
      _SAMPLE_AppInit in sample_app.c.o
  "_CFE_TBL_Manage", referenced from:
      _SAMPLE_ReportHousekeeping in sample_app.c.o
  "_CFE_TBL_Register", referenced from:
      _SAMPLE_AppInit in sample_app.c.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

stanislaw, 2020-01-12:
I have realized that the aforementioned libraries are built in the same CMake tree, not in two different trees. It is just that they are not built in a direct order. This is how I have fixed it with a hack:

```diff
diff --git a/cmake/arch_build.cmake b/cmake/arch_build.cmake
index 35b9168..2f0c29d 100644
--- a/cmake/arch_build.cmake
+++ b/cmake/arch_build.cmake
@@ -424,12 +424,18 @@ function(process_arch SYSVAR)
     endforeach(APP ${TGT${TGTID}_APPLIST})

   endforeach(TGTID ${TGTSYS_${SYSVAR}})
-
+
   # Process each app that is used on this system architecture
   foreach(APP ${TGTSYS_${SYSVAR}_APPS})
     set(APP_INSTALL_LIST ${TGTLIST_${APP}})
     message(STATUS "Building App: ${APP} install=${APP_INSTALL_LIST}")
     add_subdirectory(${${APP}_MISSION_DIR} apps/${APP})
+
+    target_link_libraries(${APP} cfe_core_default_cpu1 osal psp-pc-linux target-config-WIP)
+    target_link_options(${APP} PRIVATE -all_load)
   endforeach()
```

stanislaw, 2020-01-28:
I would like to report on another important difference that I am observing when running cFS on macOS compared to Linux.

On macOS, the apps like `sample_app.so`, if they are loaded dynamically, do not "see" the global variables such as `CFE_ES_ResetDataPtr`. Instead, they get their own copies of these symbols and therefore they cannot access the right information from the main executable.

I have reduced this difference to the following question:

`How to share a global variable between a main process and a dynamic library via a static library (macOS)?` 

On Linux this kind of sharing is easily achieved by adding a `-rdynamic` flag to the linker flags of the final executable. On macOS, adding `-rdynamic` does not have such effect.

I have created a minimal example that demonstrates this difference and opened a question on StackOverflow to have this difference tracked: [How to share a global variable between a main process and a dynamic library via a static library (macOS)?](https://stackoverflow.com/questions/59901279/how-to-share-a-global-variable-between-a-main-process-and-a-dynamic-library-via).

I have experimented with the linker flags on macOS a little a bit and so far I could not achieve the behavior which is easily done with `-rdynamic` on Linux.

My next step is to switch to the static linking of the apps as was suggested by @jphickey here: https://github.com/nasa/cFE/issues/482#issuecomment-578151848.

Also, I think I will close this issue because it does not have a specific action item attached to it.

closed 2020-01-28 by stanislaw

jphickey, 2020-01-28:
Does sample_app directly access the variable from CFE_ES?  I don't see any direct references to e.g. CFE_ES_ResetDataPtr from sample_app.

It sounds like you might have not linked the code correctly as a shared/dynamic module. Make sure you didn't inadvertently link another entire copy of CFE ES (text+data) into your app module.

stanislaw, 2020-01-28:
Thanks for asking. It accesses this variable a few calls down the stack:

```
  * frame #0: 0x00007fff618ad2c6 libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff61968bf1 libsystem_pthread.dylib`pthread_kill + 284
    frame #2: 0x00007fff618176a6 libsystem_c.dylib`abort + 127
    frame #3: 0x00007fff617e020d libsystem_c.dylib`__assert_rtn + 324
    frame #4: 0x000000010c1d3ac2 sample_app.so`CFE_ES_SysLogAppend_Unsync(LogString="2030-027-14:42:45.79172 ES SharedData Mutex Take Err Stat=0xffffffff,App=0,Func=CFE_ES_RegisterApp,Line=726\n") at cfe_es_syslog.c:222
    frame #5: 0x000000010c1d3dc1 sample_app.so`CFE_ES_SysLogWrite_Unsync(SpecStringPtr="ES SharedData Mutex Take Err Stat=0x%x,App=%d,Func=%s,Line=%d\n") at cfe_es_syslog.c:337
    frame #6: 0x000000010c1c9ad3 sample_app.so`CFE_ES_LockSharedData(FunctionName="CFE_ES_RegisterApp", LineNumber=726) at cfe_es_api.c:1795
    frame #7: 0x000000010c1ca5a0 sample_app.so`CFE_ES_RegisterApp at cfe_es_api.c:726
    frame #8: 0x000000010c1c8c2d sample_app.so`SAMPLE_AppMain at sample_app.c:53
    frame #9: 0x000000010b4b2b1e core-cpu1`OS_TaskEntryPoint(task_id=65544) at osapi-task.c:139
    frame #10: 0x000000010b4b511c core-cpu1`OS_PthreadTaskEntry(arg=0x0000000000010008) at osapi.c:433
    frame #11: 0x00007fff619662eb libsystem_pthread.dylib`_pthread_body + 126
    frame #12: 0x00007fff61969249 libsystem_pthread.dylib`_pthread_start + 66
    frame #13: 0x00007fff6196540d libsystem_pthread.dylib`thread_start + 13
```

This is the assert that I have put to the code of the `CFE_ES_SysLogAppend_Unsync` method and this assert fails (results in the call stack above).

```
    assert(CFE_ES_ResetDataPtr != 0); // crashes here

    int idx = CFE_ES_ResetDataPtr->SystemLogWriteIdx;
```

Also, I see that the `CFE_ES_ResetDataPtr` symbol is defined twice: both in the main executable and the dynamic library. I am seeing this with `nm` tool.

I am fairly certain that the `-rdynamic` is the key to understanding this difference. Currently CFS relies on the Linux behavior of it. See the SO thread to see my findings on the smallest example: https://stackoverflow.com/questions/59901279/how-to-share-a-global-variable-between-a-main-process-and-a-dynamic-library-via.

---

What else do I have to do except doing `TGT1_STATIC_APPLIST` to enable the static loading of the apps? Are there any existing instructions? I have made this change: `TGT1_STATIC_APPLIST` and I see that the apps are now `.a` files but it looks like their content is not run by the CFS.

When I get the apps to be statically linked and see them running on macOS, I will open a rather small WIP Pull Request here on CFE which will have the minimal changes that I needed to make to run CFS/CFE on macOS.

jphickey, 2020-01-28:
I am thinking the other changes introduced to get past a previous linker issue might not be correct, and you are getting two copies of the entire CFE_ES code and symbols (maybe more).  

The sample_app should only reference API calls from CFE_ES.  When you look at "nm" output on the app module file, the symbol types should be "U", not "T", and they should _NOT_ have an address.

If you see that the code is fully linked (T) then this means the linker command for your app was incorrect.

jphickey, 2020-01-28:
For example:

```
$ nm ./build/exe/cpu1/cf/sample_app.so | grep WriteToSysLog
                 U CFE_ES_WriteToSysLog
$ nm ./build/exe/cpu1/core-cpu1 | grep WriteToSysLog
00000000000118c6 T CFE_ES_WriteToSysLog
```

This is a correctly-linked example where the code is contained in "core-cpu1" (T) and referenced by sample_app.so (U).  If you have it as "T" with an address in both then that is incorrect.

stanislaw, 2020-01-28:
We are on the same page here: I do get this behavior on Linux and I see the same nm output as you showed.

However on macOS it is different:

```
$ nm ./build/exe/cpu1/cf/sample_app.so | grep WriteToSysLog
0000000000001610 T _CFE_ES_WriteToSysLog
$ nm ./build/exe/cpu1/core-cpu1 | grep WriteToSysLog
00000001000008f0 T _CFE_ES_WriteToSysLog
```

I have double-checked the linker flags that are used and I see nothing wrong about how both the apps and the executable are linked.

stanislaw, 2020-01-28:
This is the only macOS-related difference in the diff that I am having:

```
target_link_options(${APP} PRIVATE -all_load)
```

> I am thinking the other changes introduced to get past a previous linker issue might not be correct, and you are getting two copies of the entire CFE_ES code and symbols (maybe more).

I am thinking this too as one of the possibilities. I will investigate more.

jphickey, 2020-01-28:
Yes, this confirms there is definitely something wrong with your MacOS link.  It is linking the complete CFE ES library (text+data) into your sample_app.so file.  This is incorrect, and it is why you ended up with two copies.

stanislaw, 2020-01-28:
Based on my recent debugging experience, I have managed to build everything on macOS without providing the above `-all_load` flag:

I had to only add the `= 0` to the `uint32              CFE_PSP_SpacecraftId = 0;` line and this resolves the original issue of this ticket.

However, the `T` symbol is still around so the `-all_load` flag was not a problem of the second issue. (I always rebuild the new changes from scratch to make sure any of CMake cache details do not plague my findings).

```
$ nm ./build/exe/cpu1/cf/sample_app.so | grep WriteToSysLog
0000000000001c40 T _CFE_ES_WriteToSysLog
```

stanislaw, 2020-01-28:
Ok, after fighting with this more I think I have found the right macOS flag:

Doing `-undefined dynamic_lookup` gives me the Linux behavior of `-rdynamic` and this seems to resolve both of the issues that I reported here.

```cmake
  foreach(APP ${TGTSYS_${SYSVAR}_APPS})
    set(APP_INSTALL_LIST ${TGTLIST_${APP}})
    message(STATUS "Building App: ${APP} install=${APP_INSTALL_LIST}")
    add_subdirectory(${${APP}_MISSION_DIR} apps/${APP})

    set_target_properties(${APP} PROPERTIES NO_SONAME TRUE)

    # It was wrong to do either and both of these two lines.
    # target_link_libraries(${APP} cfe_core_default_cpu1 osal psp-pc-linux target-config-WIP)
    # target_link_options(${APP} PRIVATE -all_load)

    # This works
    target_link_options(${APP} PRIVATE -undefined dynamic_lookup)
  endforeach()
```

I see that the apps are `dlopen`ed and I can hit them with a CLion debugger.