nasa/cFE #493: Warnings issued when building on ubuntu-19.10

dmknutsen, 2020-01-27:
**Describe the bug**
The following warnings are issued when building on ubuntu-19.10:

/home/dan/Desktop/cFS/apps/sample_app/fsw/src/sample_app.c: In function ‘SAMPLE_ReportHousekeeping’:
/home/dan/Desktop/cFS/apps/sample_app/fsw/src/sample_app.c:365:42: warning: taking address of packed member of ‘struct <anonymous>’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  365 |     CFE_SB_TimeStampMsg((CFE_SB_Msg_t *) &Sample_AppData.SAMPLE_HkTelemetryPkt);
      |                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/dan/Desktop/cFS/apps/sample_app/fsw/src/sample_app.c:366:37: warning: taking address of packed member of ‘struct <anonymous>’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  366 |     CFE_SB_SendMsg((CFE_SB_Msg_t *) &Sample_AppData.SAMPLE_HkTelemetryPkt);
      |                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

/home/dan/Desktop/cFS/apps/ci_lab/fsw/src/ci_lab_app.c: In function ‘CI_ReportHousekeeping’:
/home/dan/Desktop/cFS/apps/ci_lab/fsw/src/ci_lab_app.c:297:5: warning: converting a packed ‘ci_hk_tlm_t’ {aka ‘struct <anonymous>’} pointer (alignment 1) to a ‘CFE_SB_Msg_t’ {aka ‘union <anonymous>’} pointer (alignment 4) may result in an unaligned pointer value [-Waddress-of-packed-member]
  297 |     CFE_SB_TimeStampMsg((CFE_SB_Msg_t *)&CI_HkTelemetryPkt);
      |     ^~~~~~~~~~~~~~~~~~~
In file included from /home/dan/Desktop/cFS/apps/ci_lab/fsw/src/ci_lab_app.c:35:
/home/dan/Desktop/cFS/apps/ci_lab/fsw/src/ci_lab_msg.h:89:9: note: defined here
   89 | typedef struct
      |         ^~~~~~
In file included from /home/dan/Desktop/cFS/cfe/fsw/cfe-core/src/inc/cfe_evs_msg.h:41,
                 from /home/dan/Desktop/cFS/cfe/fsw/cfe-core/src/inc/cfe_evs.h:42,
                 from /home/dan/Desktop/cFS/apps/ci_lab/fsw/src/ci_lab_app.h:37,
                 from /home/dan/Desktop/cFS/apps/ci_lab/fsw/src/ci_lab_app.c:32:
/home/dan/Desktop/cFS/cfe/fsw/cfe-core/src/inc/cfe_sb.h:91:13: note: defined here
   91 |     typedef union {
      |             ^~~~~
/home/dan/Desktop/cFS/apps/ci_lab/fsw/src/ci_lab_app.c:298:5: warning: converting a packed ‘ci_hk_tlm_t’ {aka ‘struct <anonymous>’} pointer (alignment 1) to a ‘CFE_SB_Msg_t’ {aka ‘union <anonymous>’} pointer (alignment 4) may result in an unaligned pointer value [-Waddress-of-packed-member]
  298 |     CFE_SB_SendMsg((CFE_SB_Msg_t *)&CI_HkTelemetryPkt);
      |     ^~~~~~~~~~~~~~
In file included from /home/dan/Desktop/cFS/apps/ci_lab/fsw/src/ci_lab_app.c:35:
/home/dan/Desktop/cFS/apps/ci_lab/fsw/src/ci_lab_msg.h:89:9: note: defined here
   89 | typedef struct
      |         ^~~~~~
In file included from /home/dan/Desktop/cFS/cfe/fsw/cfe-core/src/inc/cfe_evs_msg.h:41,
                 from /home/dan/Desktop/cFS/cfe/fsw/cfe-core/src/inc/cfe_evs.h:42,
                 from /home/dan/Desktop/cFS/apps/ci_lab/fsw/src/ci_lab_app.h:37,
                 from /home/dan/Desktop/cFS/apps/ci_lab/fsw/src/ci_lab_app.c:32:
/home/dan/Desktop/cFS/cfe/fsw/cfe-core/src/inc/cfe_sb.h:91:13: note: defined here
   91 |     typedef union {
      |             ^~~~~

**To Reproduce**
Nominal build process

**Expected behavior**
Clean build

**System observed on:**
Oracle VM VirtualBox
OS: ubuntu-19.10
Version: cFE 6.7.5.0, OSAL 5.0.5.0, PSP 1.4.3.0,

**Reporter Info**
Dan Knutsen
GSFC/NASA

dmknutsen, 2020-01-27:
[Errors_for_Isssue_493.txt](https://github.com/nasa/cFE/files/4117989/Errors_for_Isssue_493.txt)

jphickey, 2020-01-27:
This should be fixed by #313 

However - I'm curious what version of GCC this is using and if you are cross compiling or building with SIMULATION=native?  Previously, GCC would only report possible mis-alignment issues on architectures that had strict alignment requirements.  It is possible in the newest GCC versions that they might warn even if the target CPU does not need the alignment?

Either way, I do have a fix for this, but it is waiting in the wings until I have time to rebase it and submit a pull request.

dmknutsen, 2020-01-27:
I am using gcc version 9.2.1 and I am building with SIMULATION=native.

Also, I suspect you are correct with the warnings being isolated to newer GCC versions as I only see it on my ubuntu 19.10 VM (GCC version 9.2.1) and not on my ubuntu18.4.3 VM (GCC version 7.4.0).

skliper, 2020-05-19:
@dmknutsen Could you confirm the warnings have been resolved?

dmknutsen, 2020-05-20:
Yep.  Just rebuilt this morning and ran software/unit tests.  Everything looks good now.

skliper, 2020-05-20:
Duplicate of the other memory alignment/warning tickets.  Resolved.

closed 2020-05-20 by skliper